WildernessLabs/Meadow_Issues #232: Writing to the Console immediately after wake from sleep causes likely OS crash (app stops executing)

ctacke, 2022-09-23:
OS: 0.6.7.19
Hardware: FeatherV2

Repro
```
public override Task Initialize()
{

    Device.PlatformOS.AfterWake += () =>
    {
        Resolver.Log.Info("Device has returned from Sleep mode");
    };

    return Task.CompletedTask;
}

public override async Task Run()
{
    Resolver.Log.Info("Device entering Sleep mode");
    Device.PlatformOS.Sleep(TimeSpan.FromSeconds(5));

    while(true)
    {
        // these never occur
        Resolver.Log.Info("tick");
        await Task.Delay(1000);
    }
}
```

Peter-Moody, 2022-09-24:
Cause appears to be that HCOM is in the process of receiving message and when the F7 wakeup occurs the read call generates an error (errno 128 - Transport endpoint is not connected). HCOM then reconnects but by then the message has been sent.
Proposed Solution - Close all the serial read connections (USB serial, stdout and stderr) before entering Sleep Mode. This will prevent an error and the connections reconnected after the wakeup.

patridge, 2022-11-14:
Whenever this is ready to release, we should update the "known issue" details in the docs as well.
http://beta-developer.wildernesslabs.co/Meadow/Meadow_Basics/Apps/Sleep/#rc1-known-issue

Peter-Moody, 2023-07-26:
I fixed this a while back. The fix was to close the various serial port held open by HCOM before entering the Low-power mode,

Peter-Moody, 2023-07-26:
Need to verify that this has been fixed.

closed 2024-03-19 by adrianstevens