WildernessLabs/Meadow_Issues #234: Feature Request:  Please implement a USB Device stack and API

JinShil, 2022-10-17:
Please create a stack and API for building USB devices.  This would include USB-to-Serial devices, which I believe are already implemented or on the roadmap, but also other USB devices such as HID devices (e.g. keyboards, mice, etc.), audio devices, storage devices, and more.  The USB-to-Serial device would be just one implementation using the USB Device stack and API.

adrianstevens, 2023-03-30:
@acassis can you investigate?

jpearsonduluth, 2024-07-04:
Any update on this? Or if I'm in the wrong place, point me in the right direction please...

Would very much like to use the built in USB for two-way serial coms with the PC. For now I'm planning to use a [USB-to-serial module](https://a.co/d/0dpxes8l) as suggested in [this thread](https://community.wildernesslabs.co/t/how-communicate-with-my-meadow/1891/2). 

Side-note: I'd have replied with this question to linked thread, but the sign in link on https://community.wildernesslabs.co lands on a 404

alancassis, 2024-07-04:
@jpearsonduluth we don't have yet implemented it. Since the only external USB is used to communicate with the computer and to program the board, we don't have a simple way to support it. I investigated about using USB Composite, but it wasn't so simple as I thought. So we need to verify how to implement it, probably it is a feature that will require a new board. I'll discuss it in more detail with the team.