WildernessLabs/Meadow_Issues #236: Add default launch.json to Meadow.Core.Samples...

CartBlanche, 2022-10-20:
Add default launch.json files to all the Meadow.Core.Samples, so that VSCode users who download the repo, just hit [Play] to get the demo going. Just one less paper cut for them.

adrianstevens, 2022-10-20:
@CartBlanche do you have an example? This should be a pretty quick update.

CartBlanche, 2022-10-20:
@adrianstevens each sample would need a `.vscode` sub-directory and within it a launch.json with the following lines

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Meadow Deploy",
            "type": "meadow",
            "request": "launch",
            "preLaunchTask": "meadow: Build"
        },

        {
            "name": "Meadow Attach",
            "type": "mono",
            "request": "attach",
            "address": "localhost",
            "port": 4024
        },
    ]
}
```

In theory they probably wouldn't use the `attach` part unless they do `meadow debug` on the command line and then use VSCode to attach to it, but maybe some devs might find that useful. Maybe a discussion about the `attach` part for the group?

CartBlanche, 2022-10-22:
@adrianstevens I know you've been busy, but did get started on the above? If not I'll try to get a PR ready by Monday.

adrianstevens, 2023-01-09:
Sorry @CartBlanche I haven't started this one yet but it should be pretty easy. Maybe @patridge could take a look too?

patridge, 2023-01-10:
If we add this to our current single source template system, will it show up in VS2022-created solutions as well? And if so, do we even need to care?

patridge, 2023-01-10:
Sorry, I was misunderstanding the goal here. This is for samples, not templates, which already appear to have the launch.json files for the executable ones.

patridge, 2023-01-10:
@CartBlanche While I'm in there, is there anything to worry about with this config warning?

[image]

patridge, 2023-01-10:
...and PR pending for [Meadow.Foundation.Samples](https://github.com/WildernessLabs/Meadow.Foundation/pull/519) as well now.

patridge, 2023-01-12:
PRs now available for Meadow.ProjectLab, Meadow.ProjectLab.Samples, and Meadow.Project.Samples.

adrianstevens, 2023-03-30:
thx @patridge - closing issue

closed 2023-03-30 by adrianstevens