WildernessLabs/Meadow_Issues #239: Creating extended output port fails with bad error

ctacke, 2022-11-05:
**Describe the bug**
Creating a DigitalOutputPort on the Device with an IPin from a hardware extender ends up with an unhelpful `IndexOutOfBounds` exception.

**To Reproduce**
With a ProjectLab, try:
```
Device.CreateDigitalOutputPort(projLab.Mcp_2.Pins.GP0, false);
```
**Expected behavior**
An error that says "you cannot use external pins on the Device" or something like that

**Developer tools (please complete the following information as best as you can):**
 - Windows 11
 - Studio 2022
 - Meadow F7v2, projectLab 2.d
 - RC-1 software

adrianstevens, 2023-01-09:
In this case I assume Device is Meadow? I guess the question partially is, should Meadow know all peripherals that have been instantiated and then be able to do things like call extender methods to create ports?

ctacke, 2023-06-07:
This has been fixed with the IPin controller pattern and came out with 1.0.0

closed 2023-06-07 by ctacke