WildernessLabs/Meadow_Issues #241: returning dynamic variables from functions hangs the code

muenchris, 2022-11-12:
**Describe the bug**
I have a function call that returns a dynamic variable:
```
            dynamic appSettings = GetAppSettingsObject();
            if (appSettings == null) //<- it never reaches this code
            {
                return false;
            }

        static dynamic GetAppSettingsObject()
        {
            dynamic appSettings = null;
            try
            {
                Assembly configManagerAssembly = null;
                try
                {
                    configManagerAssembly = Assembly.Load("System.Configuration.ConfigurationManager, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a");
                }
                catch { 
                    //intent
                }
                if (configManagerAssembly != null)
                {
                    var configManagerType = configManagerAssembly.GetType("System.Configuration.ConfigurationManager", false);
                    if (configManagerType != null)
                    {
                        try
                        {
                            appSettings = configManagerType.GetProperty("AppSettings")?.GetValue(null);
                        }
                        catch { 
                            //intent
                        }
                    }
                }
                Console.WriteLine($"GASO 3.5 {Assembly.GetEntryAssembly()?.Location}"); //<- here its still alive
                return null; //Either this is not called or 
            }
            catch { 
                //intent
            }
            return appSettings;
        }
```
even is "null" is supposed to be returned, the function never actually returns.
**To Reproduce**
Steps to reproduce the behavior:
Compile code above and run it

**Expected behavior**
at least with null returned the code should continue

**Screenshots**

**Developer tools (please complete the following information as best as you can):**
 - Windows 11 
 - Visual Studio 2022 17.4
 - VS 2022 Tools for Meadow v0.90.9

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v1.c
 - Meadow OS version: 0.9 (RC)

bryancostanich, 2022-11-12:
Great bug report!

nealef, 2023-07-03:
I created a patch to our mono implementation that addressed dynamic calls as they didn't work at all. They should be part of the current release. Would you be able to give that a test drive?

adrianstevens, 2024-07-24:
For viz - I had to install Microsoft.CSharp in order to compile with the above code

adrianstevens, 2024-07-24:
And hit this error:

```
GASO 3.5 /meadow0/Meadow.dll
cant resolve internal call to "System.Runtime.InteropServices.Ma
rshal::IsComObject(object)" (tested without signature also)

Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll

When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated 
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
cant resolve internal call to "System.Runtime.InteropServices.Ma
rshal::IsComObject(object)" (tested without signature also)

Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll

When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated 
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
[core] App Error in App Run:  assembly:<unknown assembly> type:<unknown type> member:(null)
```

halyssonJr, 2025-03-18:
I got  this error:
```
cant resolve internal call to "System.Runtime.InteropServices.Ma
rshal::IsComObject(object)" (tested without signature also)

Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll

When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated 
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
cant resolve internal call to "System.Runtime.InteropServices.Ma
rshal::IsComObject(object)" (tested without signature also)

Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll

When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated 
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
[core] App Error in App Run:  assembly:<unknown assembly> type:<unknown type> member:(null)
[application]  System.MissingMethodException:  assembly:<unknown assembly> type:<unknown type> member:(null)
[application]   at (wrapper managed-to-native) System.Runtime.InteropServices.Marshal.IsComObject(object)
  at Microsoft.CSharp.RuntimeBinder.BinderHelper.IsComObject (System.Object obj) [0x00003] in <0fa3ab96383545bf8dfc75d27a502dc6>:0 
  at Microsoft.CSharp.RuntimeBinder.BinderHelper.IsDynamicallyTypedRuntimeProxy (System.Dynamic.DynamicMetaObject argument, Microsoft.CSharp.Runt
meBinder.CSharpArgumentInfo info) [0x00011] in <0fa3ab96383545bf8dfc75d27a502dc6>:0 
  at Microsoft.CSharp.RuntimeBinder.BinderHelper.DeduceArgumentRestriction (System.Int32 parameterIndex, Microsoft.CSharp.RuntimeBinder.ICSharpInvokeOrInvokeMemberBinder callPayload, System.Dynamic.DynamicMetaObject argument, Microsoft.CSharp.RuntimeBinder.CSharpArgumentInfo info) [0x00011] in <0f
3ab96383545bf8dfc75d27a502dc6>:0 
  at Microsoft.CSharp.RuntimeBinder.BinderHelper.Bind (Microsoft.CSharp.RuntimeBinder.ICSharpBinder action, Microsoft.CSharp.RuntimeBinder.RuntimeBinder binder, System.Dynamic.DynamicMetaObject[] args, System.Collections.
G
eneric.IEnumerable`1[T] arginfos, System.Dynamic.DynamicMetaObject onBindingError) [0x00094] in <0fa3ab96383545bf8dfc75d27a502dc6>:0 
  at Microsoft.CSharp.RuntimeBinder.CSharpUnaryOperationBinder.FallbackUnaryOperation (System.Dynamic.DynamicMetaObject target, System.Dynamic.DynamicMetaObject errorSuggestion) [0x0000b] in <0fa3ab96383545bf8dfc75d27a502dc6>:0 
  at System.Dynamic.
naryOperationBinder.FallbackUnaryOperation (System.Dynamic.DynamicMetaObject target) [0x00000] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at System.Dynamic.DynamicMetaObject.BindUnaryOperation (System.Dynamic.UnaryOperationBinder binder) [0x0000b] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at System.Dynamic.UnaryOperationBinder.Bind (System.Dynamic.DynamicMetaObject target, System
Dynamic.DynamicMetaObject[] args) [0x00020] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at System.Dynamic.DynamicMetaObjectBinder.Bind (System.Object[] args, System.Collections.ObjectModel.ReadOnlyCollection`1[T] parameters, System.Linq.Expressions.LabelTarget returnLabel) [0x000c6] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at System.Runtime.CompilerServices.CallSiteBinder.BindCor
[T] (System.Runtime.CompilerServices.CallSite`1[T] site, System.Object[] args) [0x00019] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at System.Dynamic.UpdateDelegates.UpdateAndExecute1[T0,TRet] (System.Runtime.CompilerServices.CallSite site, T0 arg0) [0x000ff] in <dfce1f887b7b4935b620c0ebec35e743>:0 
  at test.meadow_feather.MeadowApp.Run () [0x000a0] in <cb02885586df4e8abdd1e4b4e5
ce2e9>:0 
  at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x001f6] in <76f44334b0374cb4a3b83fb49afed26b>:0 

```

halyssonJr, 2025-03-31:
@muenchris, thanks for reporting this issue. This issue was solved in this PR:
- https://github.com/WildernessLabs/Meadow/pull/716

This will be available in the next release.

closed 2025-04-29 by halyssonJr