WildernessLabs/Meadow_Issues #242: With JIT enabled, reflection on objects fail on PropertyInfo.GetValue() method

muenchris, 2022-11-13:
**Describe the bug**
With the JIT compiler on, it looks like reflection is not working properly on PropertyInfo reflection.

```
    public class ThePropertyBag : List<string>
    {
        internal Dictionary<string, string> _FastCache;    
        public static ThePropertyBag Create(object pInObj)
        {
            if (pInObj == null) return new();

            ThePropertyBag tBag = new ()
            {
                _FastCache = new Dictionary<string, string>()
            };
            Type type = pInObj.GetType();
            PropertyInfo[] info = type.GetProperties();
            foreach (PropertyInfo property in info)
            {
                object Value;
                try
                {
                    Value = property.GetValue(pInObj, null); //<- here it excepts with "GetMethod not found for xxx"
                    if (Value == null) continue;
                }
                catch (Exception e)
                {
                    Console.WriteLine($"PropBag ({pInObj}) Exception: {e.Message}");
                    continue;
                }
                bool tAddVal = true;
                switch (property.PropertyType.FullName)
                {
                    case "System.String":
                        if (Value.ToString() == "") tAddVal = false;
                        break;
                    case "System.Guid":
                        if (TheCommonUtils.CGuid(Value) == Guid.Empty) tAddVal = false;
                        break;
                    case "System.Double":
                    case "System.Int32":
                    case "System.Int":
                    case "System.Int64":
                    case "System.Long":
                    case "System.Byte":
                        // ReSharper disable once CompareOfFloatsByEqualityOperator
                        if (TheCommonUtils.CDbl(Value) == 0) tAddVal = false;
                        break;
                }
                if (tAddVal)
                {
                        tBag.Add($"{property.Name}={Value}");
                        tBag._FastCache.TryAdd(property.Name, Value.ToString());
                }
            }
            return tBag;
        }
}
```

**To Reproduce**
Call the code with "ThePropertyBag.Create(some object)".
The code should create a List<string> of all Properties with "{Name}={Value}" as strings
_FastCache is used for faster lookup 

**Expected behavior**
This works fine on the desktop

**Developer tools (please complete the following information as best as you can):**
Windows 11
Visual Studio 2022 17.4
VS 2022 Tools for Meadow v0.90.9
Meadow hardware version: F7v1.c
Meadow OS version: 0.9 (RC)

adrianstevens, 2023-01-09:
Thx @muenchris - we've observed something similar and will investigate. @ctacke filled a related issue here #263

ctacke, 2023-04-10:
Attempting to reproduce this with the latest release (RC3) and it is now fixed:

```
Firmware Versions
    OS: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
    Mono: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
    Coprocessor: 0.9.6.0 built 07 Mar 2023 21:05:31 UTC (c6708b58)
    Protocol: 7
```

closed 2023-04-10 by ctacke