WildernessLabs/Meadow_Issues #244: Values returned by DriveInfo.GetDrives() are not set (zero)

doingnz, 2022-11-16:
**Describe the bug**
DriveInfo is not returning valid information on storage available in Meadow file system. Values all zero.

**To Reproduce**
Steps to reproduce the behavior:
1. See the Microsoft example on using DriveInfo to retrieve storage info here: https://learn.microsoft.com/en-us/dotnet/api/system.io.driveinfo.availablefreespace?view=net-7.0
2. Uses ```DriveInfo[] allDrives = DriveInfo.GetDrives();```

**Expected behavior**
Returns information on the space available and used in Meadow file system.

**Screenshots**
```
Output on Meadow RC1
Drive /
Drive type: Unknown
Volume label: /
File system: 
Available space to current user:              0 bytes
Total available space:                        0 bytes
Total size of drive:                          0 bytes 
```

**Developer tools:**
 - OS and version: Windows 11 V 10.0.22000 Build 22000
 - IDE and version: Visual Studio Professional 2022 v17.4.0
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.90.0

**Meadow:**
Meadow by Wilderness Labs
Board Information - Model: F7Micro, Hardware version: F7v2, Device name: MeadowF7
Hardware Information - Processor type: STM32F777IIK6, ID: 19-00-32-00-07-51-38-34-35-31-38-33, Serial number: 336A31583438, Coprocessor type: ESP32
Firmware Versions - OS: 0.9.0.2 built 22 Oct 2022 07:49:53 UTC (0xaef26433/) Mono: 0.9.0.2, built 22 Oct 2022 07:49:53 UTC (0xaef26433/), Coprocessor: 0.9.0.2
MAC Address: WiFi: 94:B9:7E:91:0D:AC

adrianstevens, 2023-03-30:
@ctacke just a bump on this next time you're in Meadow.Core

alexischr, 2024-10-29:
This should be looked into, this is fundamental functionality and useful data for the app to build paths from.

closed 2024-10-29 by alexischr