WildernessLabs/Meadow_Issues #245: JSON deserialization unusable with trimming linker enabled

Vazde, 2022-11-16:
**Describe the bug**
It seems that the new linker strips important constructors from classes, some of which are used for JSON deserialization. This seems to affect at least System.Text.Json and LitJson, both of which need them for creating new instances of objects using a parameterless constructor.

**To Reproduce**
1. Create a new project with the following code snippet:
```csharp
var json = "{\"Foo\": 123}";
var record = System.Text.Json.JsonSerializer.Deserialize<DemoClass>(json);
```

2. Create a libary project containing the class:
```csharp
public sealed class DemoClass {
    public int Foo { get; set; }
}
```

3. Observe how deserialization fails: "System.NotSupportedException: Deserialization of types without a parameterless constructor, a singular parameterized constructor, or a parameterized constructor annotated with 'JsonConstructorAttribute' is not supported."

**Expected behavior**
I expect deserialization to work.

**Developer tools:**
 - OS and version: Windows 10 21H1
 - IDE and version: VS 2022 Preview 17.5.0 Preview 1.0
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.90.0

**Meadow:**
 - Meadow hardware version: F7v1
 - Meadow OS versions: 0.9.0.2

**Additional context**
I don't personally have any experience with using trimming with .NET, but the easiest apparent solution would be to expose a setting for disabling trimming for certain assemblies. At least in my case all the affected classes are concentrated in one or two smaller client library assemblies. In the general case though it might not be as simple.

It would probably be a good idea to see how things like this are handled in Roslyn; there's a lot of attributes available there to mark things as dynamically accessed - _perhaps_ those would be of use here, too? I also really don't know if my example would work unchanged under Roslyn's trimming or not - it could very well be that the better solution would be to use the new code-generated JSON converters, which might also help to avoid this problem here. Unfortunately not all libraries have yet been updated to make use of those.

adrianstevens, 2023-03-30:
@jorgedevs we should test and see if this is still an issue

johnkattenhorn, 2023-07-12:
Maybe a workaround is to list it in the app.build.yaml from 1.0.4 and see if this resolves it @Vazde , FAO @adrianstevens

adrianstevens, 2024-03-07:
We'll be shipping an ultralight-weight JSON serialization library soon that will be the recommended approach for deserialiation

adrianstevens, 2024-03-19:
It's now possible to omit libraries from building linked via app.build.yaml - closing issue

closed 2024-03-19 by adrianstevens