WildernessLabs/Meadow_Issues #249: Create a OS Binary that Erases Flash

ctacke, 2022-12-16:
Sometimes a device can get into a bad state where flash is corrupt, and flashing a new OS doesn't allow things to move forward.  The device appears bricked.  It will not enumerate as a serial port, but in bootloader mode it *will* enumerate.  DFU cannot erase the external flash so we need a small "recovery OS" that boots, pauses (to allow subsequent OS imaging), then erases flash.  We could use it to attempt to unbrick these devices.

duduita, 2024-01-25:
I think that this issue has been already resolved. By using the [v0.0.7.0 OS](https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1700312693945659?thread_ts=1694790022.494669&cid=CQWD660E8), right @NevynUK?

NevynUK, 2024-01-25:
Correct, 0.0.7.0 allows user to do this.

Closing issue.

closed 2024-01-25 by NevynUK