WildernessLabs/Meadow_Issues #250: Where are schematics for Core Compute Dev board?

FraserElectronics, 2022-12-20:
As the title says, I cannot find the schematics for the Core Compute Dev board so am unsure what pins are connected where.

Andy

patridge, 2022-12-20:
@FraserElectronics Is this the file you might be looking for?
https://github.com/WildernessLabs/Meadow_Hardware_Designs/blob/main/Meadow_F7v2/Core-Compute_Module/F7CoreComputeModuleSchV2a.pdf

FraserElectronics, 2022-12-20:
@patridge No, that is the Core Compute Module itself. 
I purchased the [https://store.wildernesslabs.co/collections/frontpage/products/meadow-f7v2-core-compute-breakout-board](url) development kit and can't seem to find the schematic for it. 

Andy

FraserElectronics, 2022-12-20:
It is an odd board as it takes a Feather as well but there are no connectors for the Feather I/O so a bit baffled as to what use that part of the board is !

patridge, 2022-12-20:
Hmm...in that case, I'm not sure I follow what you need. Are you wanting to know what the pins are in the blue header section in this picture (borrowed from the store page)?

[image]

I have a Core-Compute module dev kit in front of me. It doesn't require a second Feather board at all. I'm fairly certain that is just headers that are equivalent with the headers on a Feather board for folks who are already familiar with that form factor and want to plug in components to the Core-Compute module similarly while they develop their own Core-Compute build plans.

The pins on those headers should be exactly as labeled except for D15, which is reserved for the SD card system ([note on this exception in the docs](https://developer.wildernesslabs.co/Meadow/Getting_Started/Getting_Started_Core-Compute_Module/#core-compute-pin-details)).

patridge, 2022-12-20:
Just as a quick example, here's me jumping an LED between pin D11 and ground (GND) so I can light it up using the `Device.Pins.D11` pin in code.

[image]

ctacke, 2022-12-20:
Those headers aren't for a Feather board, it's simply the pins from the Core Compute brought out to headers is a feather layout, so you could plug in a Feather peripheral easily if you wanted to.

The dev breakouts are here:
https://github.com/WildernessLabs/Meadow.Core-Compute.DevBreakouts

FraserElectronics, 2022-12-20:
Okay, that makes sense - I had thought I had read somewhere that the dev board also supported a Feather.
The schematic you link to is what I was looking for - many thanks 👍

Andy

closed 2022-12-20 by ctacke