WildernessLabs/Meadow_Issues #251: Population of the WiFi IP address can take a short time following connection.

NevynUK, 2022-12-31:
It can often take several iterations of the loop below before the IP address property is valid:

```c#
var wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
await wifi.Connect(Secrets.WIFI_NAME, Secrets.WIFI_PASSWORD);

while (wifi.IpAddress == System.Net.IPAddress.None)
{
    Console.WriteLine("TAG");
    await Task.Delay(500);
}

Console.WriteLine($"Connected {wifi.IpAddress}");
```

halyssonJr, 2025-02-10:
@NevynUK, After testing version `2.0.5.0`, I only waited for a single interaction to update the IP address. Also, I got the IP correctly .

```
[esp] WiFi adapter state changed to: Connecting
[esp] Sending command to connect
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 47, message ID: 0x00000002
[esp] Getting event data for message ID 0x00000002
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnecting returned CompletedOk
[application] Network connecting...
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 36, message ID: 0x00000003
[esp] Getting event data for message ID 0x00000003
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnectedEvent returned CompletedOk
[esp] WiFi adapter state changed to: Connected
[esp] SendingCommand returned: CompletedOk
TAG
[application] Interface: wlan0: wlan0 Wireless80211 Up
Connected 192.168.15.6

```

NevynUK, 2025-02-10:
In which case I suggest we close the issue.

closed 2025-02-10 by NevynUK