WildernessLabs/Meadow_Issues #252: System connection event appears to fire

NevynUK, 2022-12-31:
**Description**
The system seems to register a connection to an access point when it should have failed.

**To Reproduce**
1. Deploy an application to a board and successfully connect to an access point.
2. Modify the application to use the `wifi.config.yaml` file with invalid credentials and auto connect turned on.
3. Run the application.

alexischr, 2024-04-30:
@NevynUK is this issue still around?

halyssonJr, 2025-03-17:
@alexischr, this issue was tested and works as expected, but it seems that when the Disconnected is fired, the system is broken.

```
Initializing OS... 
Parsing app.config.yaml...
[core] Log level: Trace
[core] Looking for app assembly...
[core] Found '/meadow0/App.dll'
[core] Looking for IApp...
[core] MeadowApp
[core] Creating 'F7CoreComputeV2' instance took 883ms
[core] Device Initialize starting...
[core] Device is configured to use WiFi for the network interface
[esp] Starting Esp32Coprocessor event handler thread.
[esp] Waiting for event.
[core] Device configured to auto-connect to SSID 'PTHREAD'
[esp] Current state: Unknown
[esp] WiFi adapter state changed to: Connecting
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 47, message ID: 0x00000002
[esp] Getting event data for message ID 0x00000002
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnecting returned CompletedOk
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 47, message ID: 0x00000003
[esp] Getting event data for message ID 0x00000003
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnecting returned CompletedOk
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 46, message ID: 0x00000004
[esp] Getting event data for message ID 0x00000004
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface 1, event code: 37, message ID: 0x00000005
[esp] Getting event data for message ID 0x00000005
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent ConnectionRetryCountExceeded returned CompletedOk
[esp] Wifi InvokeEvent NetworkDisconnectedEvent returned CompletedOk
[esp] WiFi adapter state changed to: Disconnected
[application] Task ID 1 faulted: One or more errors occurred. (Index and count must refer to a location within the buffer.
Parameter name: bytes)
Inner Exception: Index and count must refer to a location within the buffer.
Parameter name: bytes
[core] PlatformOS Initialize starting...
[core] SD Card is configured to 'off'
[core] File system Initialize starting...
[core] Creating instance of MeadowApp...
[core] Creating 'MeadowApp' instance took 67ms
[core] Checking for Hardware property
[core] No Hardware property found in App.
[core] All cloud features are disabled.
[core] Initializing App
[application] Initialize...
[core] Running App
[core] App shutting down
[core] Shutdown
```

halyssonJr, 2025-03-17:
This issue was fixed.
- https://github.com/WildernessLabs/Meadow.Core/pull/621

closed 2025-03-17 by halyssonJr