WildernessLabs/Meadow_Issues #254: Private nuget repositories cause meadow os install to fail

mpinkus, 2023-01-04:
**Describe the bug**
I have 3 private corporate nuget repos, if these are enabled in VS during meadow os install it fails indicating it could not log into those repos ... should only use nuget.org ...

**To Reproduce**
Steps to reproduce the behavior:
open vs
add a private repo
login to private repo
do meadow os install

easy to work around, just uncheck the private repos in VS, do install, recheck

adrianstevens, 2023-01-04:
Interesting, I don't know if we have control over how Visual Studio prioritizes nuget repos so this may not be controllable within the Meadow extension. @CartBlanche for visibility.

adrianstevens, 2023-03-30:
@mpinkus is this still an issue? happy to take a closer look if this is blocking. And can you post the specific error message?

adrianstevens, 2024-03-26:
This is most likely a Visual Studio issue and not controllable via the Meadow extension - closing issue

closed 2024-03-26 by adrianstevens