WildernessLabs/Meadow_Issues #255: Microsoft.Azure.Devices.Client dependancy implementations

mpinkus, 2023-01-04:
**Additional context**
without this and (maybe) AWS IOT implemented, there is very little argument for enterprise use.

Or a clear, full implementation "Lite"... if every developer has manually implement cloud management/twin, messaging and direct methods you will loose them before they see any success ... 

Level of effort to accomplish?

mpinkus, 2023-01-04:
my code:

in init()
                #region Azure IOT Hubs

                azureIOTSettings = new AzureIOTSettings();

                deviceClient = DeviceClient.CreateFromConnectionString(azureIOTSettings.DeviceConnectionString, TransportType.Mqtt);

                deviceClient.SetMethodHandlerAsync("OnPrint", OnPrintHandler, null).Wait();

                #endregion Azure IOT Hubs

AzureIOT:
  DeviceConnectionString: HostName=*****    as settings...   i can provide a real connection string if anybody wants to debug it ...

adrianstevens, 2023-01-04:
Absolutely agreed - improved Azure, GCP and AWS support are very high on the priority list

rido-min, 2023-10-04:
maybe this alt. implementation might help to create a IoTHub DeviceClient for Meadow using MqttNet

https://github.com/iotmodels/MQTTnet.Client.Extensions.AzureIoT

DarkNoir, 2024-01-15:
This would be nice....

adrianstevens, 2024-04-16:
@jorgedevs let's priortize