WildernessLabs/Meadow_Issues #256: Device.Information.DeviceName does not change in time to catch an automated AP connect

mpinkus, 2023-01-04:
**Describe the bug**
When set Device.Information.DeviceName in init(), if you then look at your wifi AP post init you see the device as FeatherV2 but not the new name ... I realize you can set the device name in yaml, however that precludes setting the device name dynamically...

I am assuming because the wifi.config.yaml read and set before init() runs ... so not sure if the is a feature request or a bug ...
I would think a ?PreSettingsInit()? would be useful to be able to override externally facing properties such as device name or eventually AP SSID etc...

In my specific use case, I will be getting the device name and softAP name, password as a cloud-based setting on initial device boot ... it would be awkward to have to read yaml then reboot or something

**Expected behavior**
I would like to be able to set the device name (future softAP SSID etc) before the wifi auto connects based on the yaml settings in wifi.config.yaml   basically the wifi.config.yaml  SSID and passsword will be defaults that are overriden by dynamically received values stored on local disk

adrianstevens, 2023-01-04:
Thx @mpinkus  - @NevynUK and @ctacke for input/visibility

halyssonJr, 2025-01-29:
@mpinkus, regarding `Device.Information.DeviceName`, it's always updated when you overwrite the `meadow.config.yaml`, and also when you change the device name the SoftAp Hostname changes too. I could confirm this by testing the major release `2.0.0.3`. Now, considering your project I suggest:

Edit the `meadow.config.yaml`

```
Coprocessor:

    AutomaticallyStartNetwork: false

    AutomaticallyReconnect: false

    MaximumRetryCount: 7
```

Your code:

```
[...]
var isF7PlatformOS = Device.PlatformOS is F7PlatformOS;
await wifi.Connect()
[...]
ConnectTo TimeSpan.FromSeconds(45)
Device.Information.DeviceName = "New Name";

```