WildernessLabs/Meadow_Issues #258: Can't flash from v0.9.2.5 without manually entering bootloader mode

adrianstevens, 2023-01-06:
Flash v0.9.2.5 to an F7 Feather V2
Reset the board
Flash v0.9.2.5 again (so starting in normal mode, not boot loader mode)
the DFU OS write will complete
the board will reset
the onboard LED will turn white
the feather board will be unresponsive
after several seconds (5?) the board will reset

adrianstevens, 2023-03-30:
Here's the CLI change needed to reenable 

https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI/Commands/DeviceManagement/FlashOsCommand.cs#L59

jorgedevs, 2024-05-08:
I'm closing this issue, we're at v1.11.0 :). Feel free to re-open if it still happens today.

adrianstevens, 2024-05-08:
This is still an issue - we need to decide if it's something we're going to fix

closed 2024-10-29 by alexischr