WildernessLabs/Meadow_Issues #260: CLI file write error message is horrible

ctacke, 2023-01-08:
Attempted a `meadow file write` and it failed.  The failure message is very unfriendly and should be formatted for user consumption

```
Writing Validation.Utility.dll as Validation.Utility.dll to partition 0
Meadow rejected download request with an empty response message (DownloadStartFail)
ERROR
Meadow.CLI.Core.Exceptions.MeadowCommandException: Meadow rejected download request with an empty response message (DownloadStartFail)
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.SendTheEntireFile(FileCommand command, Boolean lastInSeries, CancellationToken cancellationToken) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDevice.Comms.cs:37
  at Meadow.CLI.Core.Devices.MeadowLocalDevice.WriteFile(String sourceFileName, String destinationFileName, TimeSpan timeout, CancellationToken cancellationToken) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI.Core\Devices\MeadowLocalDevice.FileManager.cs:154
  at Meadow.CLI.Commands.Files.WritesFileCommand.ExecuteAsync(IConsole console) in D:\a\Meadow.CLI\Meadow.CLI\main\Meadow.CLI\Commands\Files\WriteFileCommand.cs:78
  at CliFx.CliApplication.RunAsync(ApplicationSchema applicationSchema, CommandInput commandInput) in D:\a\CliFx\CliFx\CliFx\CliApplication.cs:147
  at CliFx.CliApplication.RunAsync(IReadOnlyList`1 commandLineArguments, IReadOnlyDictionary`2 environmentVariables) in D:\a\CliFx\CliFx\CliFx\CliApplication.cs:191
Done!
```

adrianstevens, 2023-01-08:
...at least the Done! at the end is friendly :)

We should definitely improve

alexischr, 2023-02-07:
This goes for errant exceptions which spew a huge stacktrace. Maybe it's time to introduce a UserFriendlyException that we always end up display. It's either a earlier exception transformed for friendliness, or a 'Internal Error" message with the exception being put in an "error.log" of sorts

thegreatco, 2023-03-21:
perhaps print just the message portion unless `--debug` or something is passed?

CartBlanche, 2023-03-31:
@ctacke Would this be a better message 

> Meadow was unable to download the file (DownloadStartFail)

The issue is that this is thrown as an exception as...

`throw new MeadowCommandException(command, $"Meadow was unable to download the file (DownloadStartFail)", response);`

And then trapped and rethrown later as...
```
            catch (Exception ex)
            {
                Logger?.LogError($"{ex.Message}");
                throw;
            }
```

Which generates the stack trace.

adrianstevens, 2024-03-26:
fixed in CLI v2.x

closed 2024-03-26 by adrianstevens