WildernessLabs/Meadow_Issues #262: Need a way to configure the linker for additional assemblies

ctacke, 2023-01-08:
FeatherV2, firmware 0.9.2.5, Windows

I have a Meadow application.  I have included an assembly (DLL) in the project and set the Copy to Output Directory to "Copy Always".  The assembly does end up in the build output properly.  It does not end up in either the `prelink_bin` or `postlink_bin` folders, and does not end up transferred to the target device.

I need a way to tell the deployment tools that I want that file deployed (and don't tree shake it).

CartBlanche, 2023-03-30:
@adrianstevens could it be that we're ignoring any assemblies that aren't linked to anything explicitly?

johnkattenhorn, 2023-07-12:
@CartBlanche, @ctacke - is this issue resolved or impacted by the work we tested together ? Maybe listing it in the build yaml should / could imply copy and don't trim ?

ctacke, 2024-12-12:
`app.build.yaml` fixes this

closed 2024-12-12 by ctacke