WildernessLabs/Meadow_Issues #263: Reflection is not finding public instance methods in a type

ctacke, 2023-01-08:
FeatherV2, firmware 0.9.2.5

Create a basic Meadow Library with a single public class containing a single public method.  Contents likely don't matter, but for completeness this is what I used:

```
namespace Validation.Utility
{
    public class TestClass
    {
        public int Add(int number1, int number2)
        {
            return number1 + number2;
        }
    }
}
```

Now deploy that assembly after deploying the application and load it via reflection

```
> meadow write file -f Validation.Utility.dll
```

Attempting to load a public method in the assembly fails - in fact it says there are no public instance methods at all in the type.  The following will yield a list of zero methods from the above type declaration.

```
var path = Path.Combine(AppDomain.CurrentDomain.BaseDirectory, "Validation.Utility.dll");
Resolver.Log.Info($"Loading assembly '{path}'...");
var assembly = Assembly.LoadFrom(path);

if (assembly == null)
{
    Resolver.Log.Error("Validation.Utility.dll assembly is null");
    return Task.FromResult(false);
}

Module? m = null;

Resolver.Log.Info("Validation.Utility.dll contains modules:");
foreach (var module in assembly.GetModules())
{
    Resolver.Log.Info($"    {module.Name}");

    if (module.Name == "Validation.Utility.dll")
    {
        m = module;
    }
}

if (m == null)
{
    Resolver.Log.Error("Did not find module 'Validation.Utility.dll'");
    return Task.FromResult(false);
}

var type = m.GetType("Validation.Utility.TestClass", true);
if (type == null)
{
    Resolver.Log.Error("Failed to load type 'TestClass'");
    return Task.FromResult(false);
}

var testClass = Activator.CreateInstance(type);
if (testClass == null)
{
    Resolver.Log.Error("Failed to create instance of 'TestClass'");
    return Task.FromResult(false);
}

var methods = m.GetMethods(BindingFlags.Instance | BindingFlags.Public);

MethodInfo? add = null;
Resolver.Log.Info("TestClass type contains methods:");
foreach (var method in methods)
{
    Resolver.Log.Info($"    {method.Name}");
    if (method.Name == "Add")
    {
        add = method;
    }
}

if (add == null)
{
    Resolver.Log.Error("Failed to find 'Add' method in 'TestClass'");
    return Task.FromResult(false);
}

```

adrianstevens, 2023-01-09:
Related or duplicate of #242

nealef, 2023-02-08:
@ctacke Do you have that test case somewhere I can grab and build?

ctacke, 2023-04-10:
Attempting to reproduce this with the latest release (RC3) and it is now fixed:

```
Firmware Versions
    OS: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
    Mono: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
    Coprocessor: 0.9.6.0 built 07 Mar 2023 21:05:31 UTC (c6708b58)
    Protocol: 7
```

closed 2023-04-10 by ctacke