WildernessLabs/Meadow_Issues #264: Visual Studio 2022 Locks up when stopping debugging

FraserElectronics, 2023-01-15:
**Describe the bug**
Built and downloaded (debug) an application to core compute module.
Decided I needed to change some code so hit the Stop button.
Visual Studio 2022 Locks up for about 1 minute before producing the error shown below:
Hit Okay to the exception and Visual Studio locks up for another 30s before returning.

Also, when using the command ```meadow device info``` comes back with 'Unable to find Meadow'.

**To Reproduce**
Can't reproduce every time but happens several times a day for me.

**Expected behavior**
The current application to stop and Visual Studio not lock up.

**Screenshots**
[image]

**Developer tools (please complete the following information as best as you can):**
 - Windows 10 x64
 - Visual Studio 2022 17.4.2
 - Meadow 0.90.4

**Meadow (please complete the following information as best as you can):**
Meadow StdInfo: Meadow successfully started MONO
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: 0.9.2.5
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 1D-00-3E-00-0D-50-4B-55-30-38-31-20
    Serial number: 206F3857554B
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D5:7D:AC

Firmware Versions
    OS: 0.9.2.5
    Mono: 0.9.2.5
    Coprocessor: 0.9.2.5

FraserElectronics, 2023-01-15:
A comment on why this issue was closed so quickly would be appreciated.

alexischr, 2023-01-16:
Sorry @FraserElectronics ! I am not sure how that happened. My account seems to bulk-closed several issues for no apparent reason.

FraserElectronics, 2023-01-16:
No problem 👍
Thought maybe I had created a duplicate issue.

FraserElectronics, 2023-02-03:
Can this be looked at ASAP - Meadow development is unusable for me as I can only get one or on a good day two build / deploy / stop / rebuild / deploy cycles before VS locks up and I have to kill it using task manager.

Andy

alexischr, 2023-02-04:
@CartBlanche can you look at this when you are ready?

@FraserElectronics We have been chasing a bug that caused all-around device instability, that could happen at any point the filesystem is accessed. The bug is fixed in release RC 2.1 (OS version 0.9.4.0). The debugger uses the filesystem at an as-needed-basis - could you try out 0.9.4.0 and see if you it helps the debugger for you?

More info on the release: https://github.com/WildernessLabs/Documentation/blob/develop/docs/Meadow/Release_Notes/Release-Candidates/index.md

doingnz, 2023-02-19:
@FraserElectronics does your application use HttpClient? I have just raised an issue with a demo application using Clima Hack Kit as sample that experiences the same symptoms that make meadow development impossible. 

The problems only occur with this sample if HttpClient is used to get the time from online service. 

If the call to get the time that uses HttpClient is commented out sanity is restored once the app is deployed. 

You may have more success deploying a new application if you reset Meadow and then quickly start the new debugging while Meadow is booting up and before it starts to run the previously deployed application.

CartBlanche, 2023-02-19:
@alexischr @FraserElectronics sorry, this slipped off the radar. Andrew, do you have time tomorrow to jump on a call??

FraserElectronics, 2023-02-20:
@doingnz No, not using HttpClient in this application.
@CartBlanche Can you give me a day or two, I have been working on non-Meadow projects for the last few weeks and need to get my hardware setup again so I can reproduce this issue. Also, I am not sure I got around to trying 0.9.4.0 as @alexischr had requested.

Andy

CartBlanche, 2023-02-20:
@FraserElectronics Sure. I'll next be free next Monday. Definitely recommend upgrading to the latest RC, if possible.

alexischr, 2023-04-05:
@FraserElectronics Our latest release, RC3 (0.9.6) fixes runtime a issue that showed up to many users as "debugging locking up". Could you upgrade to our latest and try again please?

If you are still having issues, we will probably need more information.

FraserElectronics, 2023-04-05:
@alexischr I will give it a try this week, thanks for the heads-up 👍

Andy

lamebrain, 2023-09-05:
@FraserElectronics please let us know if you have any issues. Closing for now. Thanks!

closed 2023-09-05 by lamebrain