WildernessLabs/Meadow_Issues #273: Memory creep in MeadowApp Timer events

fredmbarrett, 2023-03-03:
When executing code from within a System.Threading.Timer event task, Meadow uses memory and then does not release it at the end of the task.

**To Reproduce**
Using a WildernessLabs ProjectLab board (required), load the Github project https://github.com/fredmbarrett/Meadow.ProjectLab.SQLiteTest and deploy
Note: comment out MeadowApp.cs lines #126 and #162 (GC.Collect())
Run the program and track memory usage increase.

Within 48 hours of running, memory will be consumed to the point where the Meadow app no longer runs. Meadow may or may not report an out of memory error.

Dev environment:
VS 2022 PC Windows 11

Files below are log captures of the app running and tracking memory usage during run.

 - Meadow hardware version: [F7v2]
 - Meadow OS version: [e.g., v0.9.2.5]

**Additional context**
[Meadow.ProjectLab.SQLiteTest.log](https://github.com/WildernessLabs/Meadow_Issues/files/10886848/Meadow.ProjectLab.SQLiteTest.log)

Add any other context about the problem here. If you are working with additional external hardware/sensors, please list them here.
[Meadow.ProjectLab.SQLiteTest-70mins.log](https://github.com/WildernessLabs
[meadow-gc-collect-result-worksheet.xlsx](https://github.com/WildernessLabs/Meadow_Issues/files/10886847/meadow-gc-collect-result-worksheet.xlsx)
/Meadow_Issues/files/10886846/Meadow.ProjectLab.SQLiteTest-70mins.log)

alexischr, 2024-10-15:
@fredmbarrett A lot of work has been done on Meadow.OS regarding memory efficiency and GC operations. Is this bug still there for e.g. version 1.14/1.15?

alexischr, 2024-10-29:
@fredmbarrett will close, feel free to re-open if memory is still an issue in such a way

closed 2024-10-29 by alexischr