WildernessLabs/Meadow_Issues #274: UDP: Broadcast: Meadow:: IP , port: p stops with long running application

doingnz, 2023-03-05:
**Describe the bug**
When TRACE is enabled in YAML, a message is output to StdOut when the Meadow advertises itself on the network. At some random time after the application has started, these messages are no longer output. The application continues to execute correctly. No error messages output to StdOut. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set trace level by including the file `app.config.json` in the root of the project and copy to output with the following contents
```{
  "Logging": {
    "LogLevel": {
      "Default": "Trace"
    }
  }
}
```
2. Run application that includes credentials to logon to a WiFi network
3. Monitor the StdOut from Meadow. 

**Expected behavior**
The user application runs correctly. The UDP advertising and StdOut message continue forever 

**Screenshots**
StdOut at start of debug session (test application needs to output the Date Time
```
Meadow StdOut: Saving (32.61328125,2023/03/04 02:15:00)...SaveReading: 2023/03/04 02:15:01 Saving temperature reading to DB
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
```
Eventually stops  2023/03/04 04:39:50 - 2023/03/04 02:15:01 = 02:24:49 (I think it has run for much longer before stopping UDP StdOut message)
```
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.2.112, port: 17756
Meadow StdOut: SaveReading: 2023/03/04 04:39:50 Successfully saved to database Count=10377 GetTotalMemory=1182520
Meadow StdOut: done!
Meadow StdOut: Saving (30.30908203125,2023/03/04 04:40:21)...SaveReading: 2023/03/04 04:40:21 Saving temperature reading to DB
Meadow StdOut: SaveReading: 2023/03/04 04:40:35 Successfully saved to database Count=10378 GetTotalMemory=1182520
Meadow StdOut: done!
```
**Developer tools (please complete the following information as best as you can):**
 - Microsoft Windows [Version 10.0.22621.1265]
 - Visual Studio Enterprise 2022 PC 17.5.1
 - VS 2022 Tools for Meadow v0.90.2
Note the error occurs too if Meadow runs standalone and `meadow listen` used to monitor the StdOut

**Meadow (please complete the following information as best as you can):**
Connecting to Meadow on COM6
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 0.9.4.0
    Mono: 0.9.4.0
    Coprocessor: 0.9.4.0

Done!

alexischr, 2023-03-30:
@Peter-Moody looks like a occasion of HCOM stopping/unable to respond even when the app is still running

ctacke, 2024-09-17:
@Peter-Moody or @alexischr any thoughts or updates on this?