WildernessLabs/Meadow_Issues #275: Template for Project New in VS2022 has wrong signature for new RgbPwmLed()

doingnz, 2023-03-05:
**Describe the bug**
Doing Project new in VS2022 generates a sample application that will not build without first correcting generated sample source. Specifically need to comment out (or delete) `device: Device`
```
            onboardLed = new RgbPwmLed(//device: Device,
                redPwmPin: Device.Pins.OnboardLedRed,
                greenPwmPin: Device.Pins.OnboardLedGreen,
                bluePwmPin: Device.Pins.OnboardLedBlue,
                CommonType.CommonAnode);
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create a New Project in VS2022

**Expected behavior**
The sample code created should build and run without modification.

**Screenshots**
Incorrect Task Initialize()
```
        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");

            onboardLed = new RgbPwmLed(device: Device,
                redPwmPin: Device.Pins.OnboardLedRed,
                greenPwmPin: Device.Pins.OnboardLedGreen,
                bluePwmPin: Device.Pins.OnboardLedBlue,
                CommonType.CommonAnode);
            
            return base.Initialize();
        }
```

**Developer tools (please complete the following information as best as you can):**
 - Microsoft Windows [Version 10.0.22621.1265]
 - Visual Studio Enterprise 2022 v17.5.1
 - Meadow extension for IDE version: 0.90.2

**Meadow (please complete the following information as best as you can):**
Connecting to Meadow on COM6
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 0.9.4.0
    Mono: 0.9.4.0
    Coprocessor: 0.9.4.0

adrianstevens, 2023-03-24:
Thanks @doingnz this should be fixed now - sorry the release was spread out over a couple of days so there was a period where you could download the new OS and the tools still had the old Meadow templates

doingnz, 2023-03-24:
awesome - just tried project new with rc-2.2. builds out the box  :-)

closed 2023-03-24 by doingnz