WildernessLabs/Meadow_Issues #276: Memory leak when returning data via Maple

doingnz, 2023-03-05:
**Describe the bug**
Application returns list of JSON serializable objects to Maple to send to HTTP client in response to HTTP GET results in a memory leak. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load Clima Hack Kit sample application
2. Add following methods to MeadowApp.cs to aid uniform data output
```
        public static string TimeStamp()
        {
            return $"{System.DateTime.Now:yyyy-MM-ddTHH:mm:ss}";
        }

        public static string LogMemory(string function, string commentCount=null, int count=0, string comment = null)
        {
            return $"LogMemory, {TimeStamp(),25},{GC.GetTotalMemory(true),12},\"{function,-30}\",
            \"{(string.IsNullOrWhiteSpace(commentCount)?"":commentCount)}\",
            {(string.IsNullOrWhiteSpace(commentCount) ? "" : count.ToString())},
            \"{(string.IsNullOrWhiteSpace(comment) ? "" : comment)}\"";
         }
```
3. Add Console.WriteLine to GetTemperatureLogs() method to call above methods and output memory used.
```
        [HttpGet("/gettemperaturelogs")]
        public IActionResult GetTemperatureLogs()
        {
            LedController.Instance.SetColor(Color.Magenta);
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", comment: "Read Database ..."));
            var logs = DatabaseManager.Instance.GetAllTemperatureReadings();
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", "logs",logs.Count, "TemperatureTable"));
            var data = MapDtoTemperatureModels(logs);
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", "data", data.Count, "DTO"));
            LedController.Instance.SetColor(Color.Green);
            return new JsonResult(data);
        }

        private static List<TemperatureModel> MapDtoTemperatureModels(List<TemperatureTable> logs)
        {
            var data = new List<TemperatureModel>();
            foreach (var log in logs)
            {
                data.Add(new TemperatureModel()
                {
                    Temperature = log.TemperatureCelcius?.ToString("00"),
                    DateTime = log.DateTime.ToString("yyyy-MM-ddTHH:mm:ss.sss "),
                    TotalMemory = log.TotalMemory,
                    ID = log.ID
                });
            }

            return data;
        }
```
4. Add additional end point that returns data without calling SQLite
```
        [HttpGet("/gettemperaturedummy/{lines}")]
        public IActionResult GetTemperatureDummy(int lines)
        {
            LedController.Instance.SetColor(Color.Magenta);
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", comment: "Read Database ..."));

            var logs = DummyList(lines);
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", "logs", logs.Count, "DUMMY"));

            var data = MapDtoTemperatureModels(logs);
            Console.WriteLine(MeadowApp.LogMemory("GetTemperatureLogs", "data", data.Count, "DTO"));

            LedController.Instance.SetColor(Color.Green);
            return new JsonResult(data);
        }
[... 93 lines cut ...]
**Meadow (please complete the following information as best as you can):**
Connecting to Meadow on COM6
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 0.9.4.0
    Mono: 0.9.4.0
    Coprocessor: 0.9.4.0

Done!

**Additional context**
Uses Clima Hack Kit sample application as base.

doingnz, 2023-03-05:
You can increase the speed of the method to populate the dummy list by only calling GC.GetTotalMemory(true) once by putting the call outside the for loop, or not calling it. 
```
        private List<TemperatureTable> DummyList(int rows)
        {
            long totalMemory = GC.GetTotalMemory(true);
            List <TemperatureTable> TempList = new List<TemperatureTable >();
            for (int i = 0; i < rows; i++)
            {
                TemperatureTable dummy = new TemperatureTable() { ID = i, TemperatureCelcius = 30, 
                         DateTime = DateTime.Now, TotalMemory = totalMemory };
                TempList.Add(dummy);
            }
            return TempList;
        }
```
As noted in the initial issue, you need to add property `public long TotalMemory { get; set; }` to ` public class TemperatureTable` if returning the memory as a property (I added it to the Model saved to the database in the path that saves the temperature to have a record of the memory used at each save. However, as demonstrated by the HTTP GET endpoint that returns dummy data, the logging to SQLite and reading SQLIte don't contribute to the significant memory leak seen in Maple.

doingnz, 2023-03-05:
Note the dummy endpoint needs a parameter that set the number of records to return. e.g for 2000 records
```
http://192.168.2.112:5417/gettemperaturedummy/2000
```
Fetching 2000 records will work once. Some form of out of memory error will be raised on the next attempt...
e.g.
```
Meadow StdOut: memalign returned null when requesting 1048576 bytes (1048576 alignment) -- out of memory?
Meadow StdOut: mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error code '142'" (142)
```
Meadow will need to be restarted.

Requesting smaller number of returned records will let you do the request multiple times, till the memory is exhausted.

alexischr, 2023-12-19:
I think this needs to be re-tested on OS 1.6; preliminary tests have shown that we are much more reliable on Maple.

bryancostanich, 2024-01-09:
@halyssonJr we think this is fixed, can you validate, please?

halyssonJr, 2024-01-17:
I carried out a validation using the App provided by the user, however there was a problem with the allocation of 12k bytes and after that the application broke. I tested with OS `v1.7.0.0` and `v1.8.0.0`
```
Meadow StdOut: Log level: Information
Meadow StdOut: Device is configured to use WiFi for the network interface
Meadow StdOut: Update Service is disabled.
Meadow StdOut: Health Metrics disabled.
Listening for Meadow Console output. Press Ctrl+C to exit
Meadow StdOut: MeadowApp: Connected to network: ip=192.168.4.37 port=5417
Meadow StdOut: SetClock(01/12/2024 10:02:29);
Meadow StdOut: Listening on http://192.168.4.37:5417/
Meadow StdOut: GET : /gettemperaturelogs --> GetTemperatureLogs
Meadow StdOut: GET : /gettemperaturepage/{page} --> GetTemperaturePage
Meadow StdOut: requestHandlers.Count: 1
Meadow StdOut: Starting up Maple HTTP Request listener.
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: Saving (26.0776556776556,01/12/2024 10:02:44)...
Meadow StdOut: SaveReading: Saving temperature reading to DB
Meadow StdOut: Could not allocate 12312 bytes
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
Meadow StdOut: UDP Broadcast: Meadow::192.168.4.37, port: 17756
```

bryancostanich, 2024-04-23:
Gonna retest after several fixes in the queue land, think that this may be fixed with them.

adrianstevens, 2024-05-28:
The team is doing a lot of work optimizing memory usage - will retest and verify this issue soon

alexischr, 2024-06-11:
@halyssonJr I would like you to re-test this when you get the chance, please!

closed 2024-07-30 by bryancostanich