WildernessLabs/Meadow_Issues #277: Unable to Create IAnalogInputPort from Device.CreateAnalogInputPort

thegreatco, 2023-03-16:
**Describe the bug**
I am unable to configure an analog pin for input.

**To Reproduce**
Steps to reproduce the behavior:
1. Call `Device.CreateAnalogInputPort(Device.Pins.A0X)`

**Expected behavior**
Get an `IAnalogInputPort` back, not a `PortInUseException`

**Developer tools (please complete the following information as best as you can):**
 - Windows 11
 - VS 2022 17.6.0 Preview 2
 - Meadow extension for IDE version: 0.95.0

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 0.9.4.0
 - Meadow Mono version: 0.9.4.0
 - Meadow coprocessor version: 0.9.4.0

adrianstevens, 2023-03-20:
Are you working with nugets or local refs?

adrianstevens, 2023-03-24:
And is anything else initialized or just the analog port?

I can't repro with the latest pre-release OS. I haven't tried with 0.9.4.0 yet.

alexischr, 2023-03-30:
Closing, feel free to re-open if still a bug

closed 2023-03-30 by alexischr