WildernessLabs/Meadow_Issues #279: RgbPwmLed.StartPulse - bug in the overloaded constructor that includes the TimeSpan pulseDuration parameter

E-R-T-W, 2023-03-31:
**Describe the bug**
It's simply a copy and paste bug between overloaded constructors...
The first line "pulseDuration = TimeSpan.FromMilliseconds(600.0);" in the code for the overloaded RgbPwmLed.StartPulse constructor that includes the TimeSpan pulseDuration, should not exist! i.e. the first line in the following code should not exist!

        public void StartPulse(Color color, TimeSpan pulseDuration, float highBrightness = 1f, float lowBrightness = 0.15f)
        {
            pulseDuration = TimeSpan.FromMilliseconds(600.0);
            if (highBrightness > 1f || highBrightness <= 0f)
            {
                throw new ArgumentOutOfRangeException("highBrightness", "onBrightness must be > 0 and <= 1");
            }

            if (lowBrightness >= 1f || lowBrightness < 0f)
            {
                throw new ArgumentOutOfRangeException("lowBrightness", "lowBrightness must be >= 0 and < 1");
            }

            if (lowBrightness >= highBrightness)
            {
                throw new Exception("offBrightness must be less than onBrightness");
            }

            Color = color;
            Stop();
            animationTask = new Task(async delegate
            {
                cancellationTokenSource = new CancellationTokenSource();
                await StartPulseAsync(color, pulseDuration, highBrightness, lowBrightness, cancellationTokenSource!.Token);
            });
            animationTask!.Start();
        }

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Meadow C# Application Project
2. Modify the templated MeadowApp.cs code line 21 from "CycleColors(TimeSpan.FromMilliseconds(1000));" to "CycleColors(TimeSpan.FromMilliseconds(2400));" 
3. Modify the template MeadowApp.cs code line 61 from "onboardLed.StartPulse(color, duration / 2);" to "onboardLed.StartPulse(color, duration / 2, 1f, 0f);"
4. You'll find that the LED Pulses twice for each color i.e. 600 milliSeconds * 4 = 2400 milliSeconds, ignoring the TimeSpan pulseDuration parameter.

**Expected behavior**
Each color should pulse the LED once for the duration of the TimeSpan pulseDuration parameter.

**Screenshots**
Not a screenshot, but here are the two overloaded constructors with IDENTICAL code (except for the signatures) and it's obvious that the first line of the 2nd overloaded constructor "pulseDuration = TimeSpan.FromMilliseconds(600.0);" should not be there - simply a copy and paste thing at the time of writing the code.
        //
        // Summary:
        //     Start the Pulse animation which gradually alternates the brightness of the LED
        //     between a low and high brightness setting.
        //
        // Parameters:
        //   color:
        //
        //   highBrightness:
        //
        //   lowBrightness:
        public void StartPulse(Color color, float highBrightness = 1f, float lowBrightness = 0.15f)
        {
            TimeSpan pulseDuration = TimeSpan.FromMilliseconds(600.0);
            if (highBrightness > 1f || highBrightness <= 0f)
            {
                throw new ArgumentOutOfRangeException("highBrightness", "onBrightness must be > 0 and <= 1");
            }

            if (lowBrightness >= 1f || lowBrightness < 0f)
            {
                throw new ArgumentOutOfRangeException("lowBrightness", "lowBrightness must be >= 0 and < 1");
            }

            if (lowBrightness >= highBrightness)
            {
                throw new Exception("offBrightness must be less than onBrightness");
            }

            Color = color;
            Stop();
            animationTask = new Task(async delegate
            {
                cancellationTokenSource = new CancellationTokenSource();
                await StartPulseAsync(color, pulseDuration, highBrightness, lowBrightness, cancellationTokenSource!.Token);
            });
            animationTask!.Start();
        }

        //
        // Summary:
        //     Start the Pulse animation which gradually alternates the brightness of the LED
        //     between a low and high brightness setting, using the durations provided.
        //
        // Parameters:
        //   color:
        //
        //   pulseDuration:
        //
        //   highBrightness:
        //
        //   lowBrightness:
        public void StartPulse(Color color, TimeSpan pulseDuration, float highBrightness = 1f, float lowBrightness = 0.15f)
        {
            pulseDuration = TimeSpan.FromMilliseconds(600.0);
            if (highBrightness > 1f || highBrightness <= 0f)
            {
                throw new ArgumentOutOfRangeException("highBrightness", "onBrightness must be > 0 and <= 1");
            }

            if (lowBrightness >= 1f || lowBrightness < 0f)
            {
                throw new ArgumentOutOfRangeException("lowBrightness", "lowBrightness must be >= 0 and < 1");
            }

            if (lowBrightness >= highBrightness)
            {
                throw new Exception("offBrightness must be less than onBrightness");
            }

            Color = color;
            Stop();
            animationTask = new Task(async delegate
            {
                cancellationTokenSource = new CancellationTokenSource();
                await StartPulseAsync(color, pulseDuration, highBrightness, lowBrightness, cancellationTokenSource!.Token);
            });
            animationTask!.Start();
        }

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [Windows 10 Pro v21H2]
 - IDE and version: [Visual Studio Professional 2022 v17.5.3]
 - Meadow extension for IDE version: [VS 2022 Tools for Meadow v0.95.0]

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: [F7FeatherV2]
 - Meadow OS version: [0.9.4.0]
 - If they are different, please provide these versions as well.
   - Meadow Mono version: [0.9.4.0]
   - Meadow coprocessor version: [0.9.4.0]

**Additional context**
None - bug is obviously just a cut and paste issue.

jorgedevs, 2023-03-31:
good catch @E-R-T-W! We'll patch this up for the upcoming release :)

E-R-T-W, 2023-04-01:
Thanks @jorgedevs!

So... I'm brand new to the Meadow F7, but I've worked with PIC microControllers with ME Labs PicBasic Pro for quite a while (I'm an old VB4/5/6 coder, then VB.NET, and now C# - absolutely love that I can code a microController with C# and .NET 👍).

I realize that this is just an LED, but I'm looking at the example code and working towards controlling a 6 zone slab heating system in a very large industrial sawmill builing in Northern Canada - a Mickey Mouse 6 zone PID controller of sorts.
The implementation of the RgbPwmLed class seems rather odd to me (but could be just me as I'm new to this).

The following simply discusses the use of the new Meadow App Project Template and the use and implementation of the RgbPwmLed class (it's where every single newbie like me starts):

#1. Since the timing of the LED Pulse routines simply rely on 'await Task.Delay()', it doesn't work very well at all due to the overhead in the microController instructions. If you crank up the pulseDuration to several seconds, then what happens is that the LED pulse gets cut short, a lot short, and so it ramps up to full brightness, but then only dims 1/2 way down before it is cut off by the cancellation from the onboardLED.Stop() routine:
        void ShowColorPulse(Color color, TimeSpan duration)
        {
            onboardLed.StartPulse(color, duration / 2);
            Thread.Sleep(duration);
            onboardLed.Stop();
        }

        public void Stop()
        {
            cancellationTokenSource?.Cancel();
            IsOn = false;
        }

#2. This implementation can do nothing more than pulse the LEDs, it's analagous to locking up the UI thread in a normal C# application. i.e. the "doing more work" in the Console.WriteLine() for() loop will NEVER run, in the following code:
        public override Task Run()
        {
            Console.WriteLine("Run...");

            CycleColors(TimeSpan.FromMilliseconds(5000));
            for (int i = 0; i < 300; i++)
            {
                Console.WriteLine("This will never run :( -> I'm doing more work now while LEDs are blinking :) for " + i + "seconds.");
                Thread.Sleep(1000);
            }
            return base.Run();
        }

Since the Meadow F7 has an onboard clock, why not implement the ShowColorPulse() routine as follows, for example:
        private void ShowColorPulse(Color color, TimeSpan duration)
        {
            DateTime dateTime = DateTime.Now;
            int interval = 25;
            float decimalPercent;
            double elapsedTime;

            onboardLed.SetColor(color, 0.01f);
            Console.WriteLine("0.01");
            Thread.Sleep(interval);
            do
            {
                elapsedTime = (DateTime.Now - dateTime).TotalMilliseconds;
                decimalPercent = (float)(elapsedTime / (duration.TotalMilliseconds / 2.0));
                if (decimalPercent > 1f) decimalPercent = 1;
                onboardLed.SetColor(color, decimalPercent);
                Console.WriteLine(((double)decimalPercent).ToString("0.00"));
                Thread.Sleep(interval);
            } while (decimalPercent < 1f);

            do
            {
                elapsedTime = (DateTime.Now - dateTime).TotalMilliseconds;
                decimalPercent = 1f - (float)((elapsedTime - (duration.TotalMilliseconds / 2.0)) / (duration.TotalMilliseconds / 2.0));
                if (decimalPercent < 0f) decimalPercent = 0f;
                onboardLed.SetColor(color, decimalPercent);
                Console.WriteLine(((double)decimalPercent).ToString("0.00"));
                Thread.Sleep(interval);
            } while (decimalPercent > 0f);

            onboardLed.Stop();
        }
[... 341 lines cut ...]
2>[2023-03-31 13:54:45] Skipping file (hash match): YamlDotNet.dll
2>[2023-03-31 13:54:45] Skipping file (hash match): System.Threading.Tasks.Extensions.dll
2>[2023-03-31 13:54:45] App.dll deploy complete  
2>[2023-03-31 13:54:45] Meadow StdInfo: Mono is disabled
2>[2023-03-31 13:54:45] Mono is disabled         
2>[2023-03-31 13:54:45] Meadow StdInfo: Mono has been enabled - restarting Meadow
2>[2023-03-31 13:54:45] Mono has been enabled - restarting Meadow
2>[2023-03-31 13:54:47] Connecting to Meadow on COM7
2>[2023-03-31 13:54:47] Meadow StdInfo: Meadow successfully started MONO
2>[2023-03-31 13:54:49] Meadow StdInfo: Mono is enabled
2>[2023-03-31 13:54:49] Mono is enabled          
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
========== Build started at 1:54 PM and took 31.880 seconds ==========
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========
========== Deploy started at 1:54 PM and took 31.880 seconds ==========

Output Window: Meadow

Launching application...

Connecting to Meadow on COM7
Connecting to Meadow on COM7
Meadow StdInfo: Meadow successfully started MONO
Listening for Visual Studio to connect on 127.0.0.1:55898
Visual Studio has Connected

E-R-T-W, 2023-04-02:
I just realized, in the example code I provided that uses the Real Time Clock, I didn't include the highBrightness and lowBrightness parameters. The example code should have been as follows:

```
        private void ShowColorPulse(Color color, TimeSpan duration, float highBrightness, float lowBrightness)
        {
            if (highBrightness > 1f) highBrightness = 1f;
            if (lowBrightness < 0f) lowBrightness = 0f;
            if (lowBrightness > highBrightness) highBrightness = lowBrightness;

            DateTime dateTimeStart = DateTime.Now;
            int interval = 25;
            float decimalPercent;
            double elapsedMilliSecs;

            onboardLed.SetColor(color, lowBrightness);
            Console.WriteLine(lowBrightness.ToString("0.00"));
            Thread.Sleep(interval);
            do
            {
                elapsedMilliSecs = (DateTime.Now - dateTimeStart).TotalMilliseconds;
                decimalPercent = (((float)(elapsedMilliSecs / (duration.TotalMilliseconds / 2.0))) * (highBrightness - lowBrightness)) + lowBrightness;
                if (decimalPercent > highBrightness) decimalPercent = highBrightness;
                onboardLed.SetColor(color, decimalPercent);
                Console.WriteLine((decimalPercent).ToString("0.00"));
                Thread.Sleep(interval);
            } while (elapsedMilliSecs <= duration.TotalMilliseconds / 2.0);

            do
            {
                elapsedMilliSecs = (DateTime.Now - dateTimeStart).TotalMilliseconds;
                decimalPercent = ((1f - (float)((elapsedMilliSecs - (duration.TotalMilliseconds / 2.0)) / (duration.TotalMilliseconds / 2.0))) * (highBrightness - lowBrightness)) + lowBrightness;
                if (decimalPercent < lowBrightness) decimalPercent = lowBrightness;
                onboardLed.SetColor(color, decimalPercent);
                Console.WriteLine((decimalPercent).ToString("0.00"));
                Thread.Sleep(interval);
            } while (elapsedMilliSecs <= duration.TotalMilliseconds);

            onboardLed.Stop();
        }
```

and it can be called as follows:

```
        public override Task Run()
        {
            Console.WriteLine("Run...");

            Task.Run(() => CycleColors(TimeSpan.FromMilliseconds(2000)));
            for (int i = 0; i < 300; i++)
            {
                Console.WriteLine("I'm doing more work now while LEDs are blinking :) for " + i + " seconds.");
                Thread.Sleep(1000);
            }
            return base.Run();
        }

        Task CycleColors(TimeSpan duration)
        {
            Console.WriteLine("Cycle colors...");

            while (true)
            {
                ShowColorPulse(Color.Red, duration, 0.5f, 0.02f);
                ShowColorPulse(Color.Green, duration, 0.5f, 0.02f);
                ShowColorPulse(Color.Blue, duration, 0.5f, 0.02f);
            }
        }
```

closed 2023-04-14 by E-R-T-W