WildernessLabs/Meadow_Issues #280: Deploy to In Circuit Debugging Hangs

E-R-T-W, 2023-04-01:
**Describe the bug**
Half the time, when deploying any app to the Meadow F7v2, everything simply hanges/freezes/locks up. Then have to reseat the Meadow USB cable and also kill VS2022 and start all over. Same thing for VS2019.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a brand new Meadow C# Application Project
2. Click on the Debug Run Button
3. Output Window:Build shows Build Successfull.
4. Output Window:Meadow shows VS has Connected, but then everything hangs, have to reseat Meadow USB cable and kill VS2022 and start all over (same thing with VS2019).

**Expected behavior**
100% of the time, deployment to in circuit debugging should not hang.

**Screenshots**
Here is the Output Window of the Build, followed by the Output Window of the Meadow, post build:
Output Window: Build

Build started...
1>------ Build started: Project: HellowMeadow, Configuration: Debug Any CPU ------
1>Skipping analyzers to speed up the build. You can execute 'Build' or 'Rebuild' command to run analyzers.
1>HellowMeadow -> D:\Data_Weyco\VS2022\Meadow\HelloMeadow\HellowMeadow sync into async\HellowMeadow\bin\Debug\netstandard2.1\App.dll
2>------ Deploy started: Project: HellowMeadow, Configuration: Debug Any CPU ------
2>[2023-03-31 13:54:20] Connecting to Meadow on COM7
2>[2023-03-31 13:54:21] Downloading version file for Meadow OS 0.9.4.0
2>[2023-03-31 13:54:21] Downloaded 285 bytes     
2>[2023-03-31 13:54:21] Meadow OS version 0.9.4.0 is already downloaded
2>[2023-03-31 13:54:21] Meadow StdInfo: Mono is enabled
2>[2023-03-31 13:54:21] Mono is enabled          
2>[2023-03-31 13:54:21] Meadow StdInfo: Mono has been disabled - restarting Meadow
2>[2023-03-31 13:54:21] Mono has been disabled - restarting Meadow
2>[2023-03-31 13:54:25] Connecting to Meadow on COM7
2>[2023-03-31 13:54:25] Meadow StdInfo: Mono is disabled
2>[2023-03-31 13:54:27] Meadow StdInfo: Mono is disabled
2>[2023-03-31 13:54:27] Mono is disabled         
2>[2023-03-31 13:54:27] Meadow StdInfo: dev/     
2>[2023-03-31 13:54:27] dev/                     
2>[2023-03-31 13:54:27] Meadow StdInfo: little0 [block]
2>[2023-03-31 13:54:27]  little0 [block]         
2>[2023-03-31 13:54:27] Meadow StdInfo: little0p0 [block]
2>[2023-03-31 13:54:27]  little0p0 [block]       
2>[2023-03-31 13:54:27] Meadow StdInfo: monostderr [char]
2>[2023-03-31 13:54:27]  monostderr [char]       
2>[2023-03-31 13:54:27] Meadow StdInfo: monostdout [char]
2>[2023-03-31 13:54:27]  monostdout [char]       
2>[2023-03-31 13:54:27] Meadow StdInfo: mtdblock0 [block]
2>[2023-03-31 13:54:27]  mtdblock0 [block]       
2>[2023-03-31 13:54:27] Meadow StdInfo: null [char]
2>[2023-03-31 13:54:27]  null [char]             
2>[2023-03-31 13:54:27] Meadow StdInfo: nxupd [char]
2>[2023-03-31 13:54:27]  nxupd [char]            
2>[2023-03-31 13:54:27] Meadow StdInfo: pwm0 [char]
2>[2023-03-31 13:54:27]  pwm0 [char]             
2>[2023-03-31 13:54:27] Meadow StdInfo: ramlog [char]
2>[2023-03-31 13:54:27]  ramlog [char]           
2>[2023-03-31 13:54:27] Meadow StdInfo: ttyACM0 [char]
2>[2023-03-31 13:54:27]  ttyACM0 [char]          
2>[2023-03-31 13:54:27] Meadow StdInfo: ttyS0 [char]
2>[2023-03-31 13:54:27]  ttyS0 [char]            
2>[2023-03-31 13:54:27] Meadow StdInfo: ttyS1 [char]
2>[2023-03-31 13:54:27]  ttyS1 [char]            
2>[2023-03-31 13:54:27] Meadow StdInfo: ttyS2 [char]
2>[2023-03-31 13:54:27]  ttyS2 [char]            
2>[2023-03-31 13:54:27] Meadow StdInfo: upd [char]
2>[2023-03-31 13:54:27]  upd [char]              
2>[2023-03-31 13:54:27] Meadow StdInfo: urandom [char]
2>[2023-03-31 13:54:27]  urandom [char]          
2>[2023-03-31 13:54:27] Meadow StdInfo: usrsock [char]
2>[2023-03-31 13:54:27]  usrsock [char]          
2>[2023-03-31 13:54:27] Meadow StdInfo: meadow0/ 
2>[2023-03-31 13:54:27] meadow0/                 
[... 276 lines cut ...]
Connecting to Meadow on COM7
Meadow StdInfo: Meadow successfully started MONO
Listening for Visual Studio to connect on 127.0.0.1:55898
Visual Studio has Connected

At that point everything is loked up :(

**Developer tools (please complete the following information as best as you can):**

    OS and version: [Windows 10 Pro v21H2]
    IDE and version: [Visual Studio Professional 2022 v17.5.3]
    Meadow extension for IDE version: [VS 2022 Tools for Meadow v0.95.0]

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running meadow device info using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).

    Meadow hardware version: [F7FeatherV2]
    Meadow OS version: [0.9.4.0]
    If they are different, please provide these versions as well.
        Meadow Mono version: [0.9.4.0]
        Meadow coprocessor version: [0.9.4.0]

**Additional context**
This happens about 1/2 the time, even when simply deploying a brand new Meadow App Project Template in both VS2022 and VS2019. Super annoying, Super time consuming!
Thanks for your help!! :)

CartBlanche, 2023-04-01:
@E-R-T-W if can take over 20 seconds for the debugger to connect to the device. Here's a small guide of the lights changing state...
• Green is Bootloader has started 
• Blue is OS started and Mono not running.
• None is Mono running (assuming not bricked) and your app starting.

Could you please let me know the full device info, as well as the meadow CLI version.

Device info: `meadow device info`
CLI version: `meadow --version` 

Having said that, we've made some more stability improvements to debugging in RC3, which is due out very soon.

E-R-T-W, 2023-04-01:
Interesting...

This morning when I plugged in the Meadow (USB cable), NOTHING!
I'm a newbie to Meadow, just got the thing, I've connected nothing to it yet other than the USB cable, I've only been playing with the onboard LEDs and such, figuring out the whole Meadow ecosystem.

Looks like I got a dud?!?

I purchased 2 Meadow F7v2's and a Core-Compute with Dual Ethernet (so far). I'll play with the first one some more and get the other 2 up and running and tested and report back to you, but it appears the first Meadow F7v2 is bricked after only 2 days of having nothing plugged into it except a USB cable :(  

I've also NEVER noticed the LED as you describe:
  _• Green is Bootloader has started
  • Blue is OS started and Mono not running.
  • None is Mono running (assuming not bricked) and your app starting._

Perhaps that's the root of the lock ups, a bad microController.

I'll let you know, and answer your questions too...

Thanks for your help!!!!! :) VERY much appreciated!!!

E-R-T-W, 2023-04-01:
Oh, also, to support the dud theory... Over the past 2 days, I noticed that after leaving the Meadow plugged in and running a blinking LED app, when I came back to it an hour or two later, the Meadow was dead, but reseating the USB cable always brought it back to life. No such luck this time :(

E-R-T-W, 2023-04-01:
Ok, I got to the bottom of the issues with the first Meadow F7 Feather v2.d...

With both my Meadow F7v2.d Feathers running the exact same code, on the 1st Meadow both the STM32F777IIK6 microController and the is42S16160J-7bli RAM chip get red hot, so hot that they will burn your finger, unlike the second Meadow, that barely even gets warm.
When the microController gets hot enough, for long enough, it shuts itself down. Once it's cooled enough, then it starts to work again, but the process repeats itself. 
Bad luck for a Meadow newbie I guess, with his very first ever Meadow Feather - I've been fighting with this thing for two days assuming the Meadow product was SUPER BETA LOL

**### Warranty**
I'm used to $5 PIC microControllers. 
**Do you guys have warranty** for a $55 USD = $75 CAD microController? (plus I had to pay huge import and duty fees too).
Do you want it back to do some forensic analysis? Or is it just not worth your while.

### **So now let's discuss the working/second Meadow F7v2.d**

The LEDs do function as you described!

With the second Meadow F7v2, the debugger still hangs over 40% of the time (9 out of 21 times), and it's definitely not due to waiting long enough for the Meadow to boot up.
The issue is isolated to the debugger - if I simply deploy the app without debugging, it succeeded 20 times in a row.
It would be nice if when deploying with the debugger, the Output Window switched from Build to Meadow, just like what happens when deploying without debugging, vs switching from Build to Debug when attaching with the debugger.

I've included the info you were asking for, for both my Meadow Feathers below:

**### Full Device Info (for the 1st Meadow F7 Feather v2.d):
Device info: meadow device info
AND
CLI version: meadow --version**

C:\Windows\system32>meadow device info
Connecting to Meadow on COM7
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 21-00-30-00-07-50-30-52-54-31-35-20
    Serial number: 2065317F5230
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D6:CC:D4

Firmware Versions
    OS: 0.9.4.0
    Mono: 0.9.4.0
    Coprocessor: 0.9.4.0

Done!

C:\Windows\system32>meadow --version
v0.90.4
Done!

**### Full Device Info (for the 2nd Meadow F7 Feather v2.d):
Device info: meadow device info
AND
CLI version: meadow --version**

C:\Windows\system32>meadow device info
Connecting to Meadow on COM9
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 1E-00-2D-00-07-50-30-52-54-31-35-20
    Serial number: 2062317C5230
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D6:D5:08

Firmware Versions
    OS: 0.9.4.0
    Mono: 0.9.4.0
    Coprocessor: 0.9.4.0

Done!

C:\Windows\system32>meadow --version
v0.90.4
Done!

I also tried to get my Core-Compute Dev with Dual Ethernet running, **but no joy!** It doesn't seem to enter into DFU Bootloader mode. I even uninstalled/removed a COM port since my second Meadow Feather was using port 9, I was thinking that perhaps it couldn't go into a 2 digit COM port, but still no joy. 
When I connect the USB cable, I get a green LED and that's it, no matter whether plug into DFU Bootloader or not, and the CLI port list command, as well as device manager, lists no COM ports.
C:\Windows\system32>meadow list ports
Done!

HELP!

Thanks for the help so far and for further help, in advance!!

E-R-T-W, 2023-04-01:
### wrt Core-Compute Dev with Dual Ethernet
I believe I got it working. After deleting virtual COM ports and trying to connect the Core-Compute into DFU Bootloader, and still no luck getting a COM port, I simply executed:
meadow flash os
and the Meadow CLI found the Core-Compute Dev and flashed it.
I now have a COM port for the Core-Compute Dev.

My version of the board is v2a. It has an onboard RGB LED on it. I'm looking for a schematic of v2a of the board. The only thing I can find is schematics of V1.d in the following link:
https://github.com/WildernessLabs/Meadow.Core-Compute.DevBreakouts/blob/main/Source/Hardware/Dev_Breakout_w_Dual-Ethernet/Schematic_v1.d.pdf

Where can I get an updated schematic for v2a that shows the connections to the RGB LED?

Thanks!

E-R-T-W, 2023-04-01:
The docs and jpg etc. is also for version 1.c of the board:
https://github.com/WildernessLabs/Hardware_Addon_Modules/blob/main/Docs/Core-Compute_Dev_Module.jpg

Where are the docs for v2.a?

Thanks!

CartBlanche, 2023-04-01:
@E-R-T-W there is a known issue with us triggering bootloader mode via software. 
A more reliable way to enter bootloader mode is to:
1. Unplug the meadow
2. Press the boot button, while plugging the meadow back in.
3. No lights will be shown in bootloader mode
4. If you look at the list of attached usb devices on your machine, you should see a device named `STM32 BOOTLOADER`

You should then be able to more reliably ` meadow flash os` your device.

CartBlanche, 2023-04-01:
@E-R-T-W I've asked others from the team to respond to your CCM and jpeg questions.

E-R-T-W, 2023-04-01:
Thanks @CartBlanche!

Sorry, what is CCM? Warranty for the flaky Meadow Feather?

By tracing out the board, I figured out (I think) that that RGB LED is connected to D09 on pin 29, D10 on pin 30, and D11 on pin 32. Now I'm debugging to figure out which pin goes to which color.

Same Lockup issue 40% to 50% of the time when deploying with Debugger.

Also, after several successful deployments and working out which pin to which color RGB LED, I got the following error:
[image]

HELP!

Thanks!

E-R-T-W, 2023-04-02:
Hello  @CartBlanche,

I'll preface my comments with the fact that I'm a Professional Electrical Engineer with 35 years of experience.

I realize that the Wilderness Labs Meadow Team is only at release candidate, but besides the software issues and bugs, there are some serious manufacturing quality control issues!!!

I've identified the problems with both my first Meadow Feather board, as well as my Core-Compute Dev board.

**1. First Meadow Feather Board:** As described already, the first of two Meadow Feather boards is generating some SERIOUS heat. Upon further investigation, the 3.3V to GND is nearly a dead short on that board (unlike my second Meadow Feather Board).
**2. Core-Compute Dev Board:** I traced out the RGB LED on the Core-Compute board and the problem is that the RGB LED was mounted 90 degrees in the wrong direction clockwise. i.e. it would work correctly if you rotated the RGB LED 90 degrees counter clockwise from how it is currently soldered to the board. Then you would indeed have the Common Annode at 3.3V, Pin 29=D09 would be the Red Cathode, Pin 30=D10 would be the Green Cathode. and Pin 32=D11 would be the Blue Cathode. 

**I'd like warranty replacements please.** I'll send both boards back to you if you want to confirm, but you'll find EXACTLY what I'm telling you. Can you please make that happen?!!?

P.S. don't mean to complain - but I have every right to - I can't imagine a true newbie dealing with this. Software bugs is one thing, but hardware bugs too?!? 🙄😮

E-R-T-W, 2023-04-02:
Hi @CartBlanche,

This has turned into so many issues in one post that I'll seperate them, in order to better track them, as well as help other users who are looking for answers to similar problems...

CartBlanche, 2023-04-02:
@E-R-T-W yes, please separate out the issues. @bryancostanich can we organise replacement boards for him.

CartBlanche, 2023-04-02:
CCM - Core Compute Module

E-R-T-W, 2023-04-02:
Excellent! Thank you @CartBlanche and @bryancostanich!

Please let me know how I can help to get the Feather and the CCM shipped out ASAP!

E-R-T-W, 2023-04-04:
Hi @CartBlanche,
Upgraded to the latest RC-3 (v0.9.6.3) this evening.
Debugger Hang Issues Solved 🥇 
Thank you!!!

CartBlanche, 2023-04-04:
Great to hear you're making progress.

E-R-T-W, 2023-04-14:
@CartBlanche I'm closing this issues since this has been resolved with the latest release RC-3 (v0.9.6.3).
Thank you - such a nice experience now that the debugger doesn't hang :)

closed 2023-04-14 by E-R-T-W