WildernessLabs/Meadow_Issues #283: Core-Compute Dev (Dual Ethernet) v2.a Lack of Documentation and no Schematics, in particular for onboard RGB LED

E-R-T-W, 2023-04-02:
**Describe the bug**
There doesn't seem to be any documentation or schematics available for the Core-Compute Dev (Dual Ethernet) v2.a.
There does exist documentation and schematics for the Core-Compute Dev v1.c and 1.d at:
https://github.com/WildernessLabs/Meadow.Core-Compute.DevBreakouts/blob/main/Source/Hardware/Dev_Breakout_w_Dual-Ethernet/Schematic_v1.d.pdf
and
https://github.com/WildernessLabs/Hardware_Addon_Modules/blob/main/Docs/Core-Compute_Dev_Module.jpg
for example, but nothing for the v2.a board.
In particular, the V2.a board has an onboard RGB LED, but no documentation or schematics as to whether the RGB LED is common anode or common cathode, and no documentation or schematics as to what pins the LEDs are connected to and whether there are inline resistors or not. 

**To Reproduce**
Steps to reproduce the behavior:
1. Searched online and could only find docs/drawings for v1.c and 1.d which don't include an onboard RGB LED.

**Expected behavior**
All board versions should have schematics/drawings and documentation.

**Screenshots**
None.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [Windows 10 Pro v21H2]
 - IDE and version: [Visual Studio Professional 2022 v17.5.3]
 - Meadow extension for IDE version: [VS 2022 Tools for Meadow v0.95.0]

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: [F7CoreComputeV2]
 - Meadow OS version: [0.9.4.0]
 - If they are different, please provide these versions as well.
   - Meadow Mono version: [0.9.4.0]
   - Meadow coprocessor version: [0.9.4.0]

**Additional context**
Please provide documentation for the Core-Compute Dev (Dual Ethernet) v2.a board.

Thanks!.

CartBlanche, 2023-04-02:
@E-R-T-W I've just chatted with Bryan. Could you please send an email to hello@wildernesslabs.co outlining the issues you've had with your boards and we'll organise to get you the newer revision of our boards.

E-R-T-W, 2023-04-02:
@CartBlanche / @bryancostanich,

Thank you! Much appreciated! I'll get the email sent out explaining issue details, and address, etc.

closed 2023-09-08 by E-R-T-W