WildernessLabs/Meadow_Issues #286: System.Net.Http.HttpClient.GetAsync() Memory Leak

E-R-T-W, 2023-04-08:
**Describe the bug**
System.Net.Http.HttpClient.GetAsync() memory leak - multiple calls (10 of them) will crash the Meadow with the following error:

> memalign returned null when requesting 1048576 bytes (1048576 alignment) -- out of memory?

But there's a work around...

This may be related to issue #269 and/or #276. I saw mention that #269 is fixed in RC-3 (v0.9.6.3), however my particular issue with System.Net.Http.HttpClient.GetAsync() Memory Leak still persists.

**To Reproduce**
I've dummied down the code and made everything essentially synchronous in order to troubleshoot the problem and make it reproducable. (The actual code extracts the Temperature from the Environment Canada Web Site).
1. Run the following code, and it will crash, every time on the 10th call to System.Net.Http.HttpClient.GetAsync(), with an out of memory error (assuming the web page hasn't significanlty changed it's content i.e. could perhaps take 11 or 12 repeated calls).
```
    public class MeadowApp : App<F7FeatherV2>
    {
        private const string WIFI_NAME = "[SSID]";
        private const string WIFI_PASSWORD = "[Password]";
        private IWiFiNetworkAdapter wifi;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.Connect(WIFI_NAME, WIFI_PASSWORD, TimeSpan.FromSeconds(60)).Wait();
            return base.Initialize();
        }

        public override Task Run()
        {
            Console.WriteLine("Run...");
            long i = 0;
            Task<HttpResponseMessage> responseTask;
            HttpResponseMessage response;
            System.Net.Http.HttpClient client;
            Console.WriteLine("pre HTTP requests");
            while (true)
            {
                client = new System.Net.Http.HttpClient();
                responseTask = client.GetAsync("https://weather.gc.ca/city/pages/ab-31_metric_e.html");
                responseTask.Wait();
                response = responseTask.Result;
                //response = null;
                i++;
                Console.WriteLine("post HTTP request #" + i);
            }
            return base.Run();
        }
    }
```
2. Work around - remove the comment from the code:
`//response = null;`
i.e. if you set response to null after every call/prior to the next call to System.Net.Http.HttpClient.GetAsync(), then problem solved.
3. The error, after 10th call to System.Net.Http.HttpClient.GetAsync() is:

> memalign returned null when requesting 1048576 bytes (1048576 alignment) -- out of memory?

**Expected behavior**
When using the exact same variable (not creating multiple new variables), one shouldn't have to null a variable between usages.
Note: the above code works without having to null response in normal console applications i.e. no memory leak using the same code in console applications.

**Screenshots**
The output, including the error is as follows:
```
Initialize...
Run...
pre HTTP requests
post HTTP request #1
post HTTP request #2
post HTTP request #3
post HTTP request #4
post HTTP request #5
post HTTP request #6
post HTTP request #7
post HTTP request #8
post HTTP request #9
memalign returned null when requesting 1048576 bytes (1048576 al
ignment) -- out of memory?
```

Then when the Meadow continues to run, after the 'out of memory' error, you repeatedly get the following error:
```
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
```

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 10 Pro v21H2
 - IDE and version: Visual Studio Professional 2022 v17.5.3
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.96.2]

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
   - Meadow coprocessor version: 0.9.6.0 built 07 Mar 2023 21:05:31 UTC (c6708b58)

**Additional context**
None.

CartBlanche, 2023-04-08:
@alexischr are you aware of this in the runtime?

E-R-T-W, 2023-04-08:
Let me check... (I assume you mean deploy vs debug...)
Hmm, let me think... How am I going to easily confirm that without outputing to the console window?

E-R-T-W, 2023-04-08:
Never mind, it's getting late here in Canada. I'm testing Release using Deploy, but still outputing to console windows...

E-R-T-W, 2023-04-08:
@alexischr, that was a very good quesition!!!

To be clear and specific, there are 4 scenarios:

1. Debug Compile with Debugger attached
2. Debug Compile with No Debugger (runtime)
3. Release Compile with Debugger attached
4. Release Compile with No Debugger attached (runtime)

The **only** scenario (the most important one in my world right now) with the "memory leak" issue is 1 i.e. Debug Compilation with Debugger attached. It's a Debugger issue. Well at least there's a work around until your team resolves this one.

Thanks for your insight and help!

E-R-T-W, 2023-04-08:
Update: scenario 2 with Debug Compile and No Debugger runs out of memory too, even with nulling all the variables, it just takes longer (40 calls to System.Net.Http.HttpClient.GetAsync()), as shown in output below:
```
07:33:48:376	post HTTP request #40
07:33:52:163	memalign returned null when requesting 1048576 bytes (1048576 al
07:33:52:163	ignment) -- out of memory?

```

Also, when trying to see if eventually scenaio 4 will fail, (Release Compile with No Debugger attached), I get the following error after 217 calls to System.Net.Http.HttpClient.GetAsync():
```
08:23:59:642	post HTTP request #217
08:24:01:760	App Error in App Run: One or more errors occurred. (The SSL connection could not be established, see inner exception.)
08:24:01:766	System.AggregateException: One or more errors occurred. (The SSL connection could not be established, see inner exception.)
08:24:02:756	at System.Threading.Tasks.Task.ThrowIfExceptional (System.Boolean includeTaskCanceledExceptions) [0x00011] in <77563d7372d948198f425b41a9b84c90>:0
08:24:02:757	at System.Threading.Tasks.Task.Wait (System.Int32 millisecondsTimeout, System.Threading.CancellationToken cancellationToken) [0x00043] in <77563d7372d948198f425b41a9b84c90>:0
08:24:02:757	at System.Threading.Tasks.Task.Wait () [0x00000] in <
08:24:02:805	7563d7372d948198f425b41a9b84c90>:0
08:24:02:805	at HttpClient.MeadowApp.Run () [0x00028] in <fdc73863f7fc46388a7a441fe3c612d1>:0
08:24:02:805	at Meadow.MeadowOS.Start (System.String[] args) [0x0010b] in <b233fa841cb445079c08d536686ee21c>:0
08:24:02:805	Inner System.Net.Http.HttpRequestException: TLS initialization or handshake failed
08:24:02:919	at System.Net.Http.ConnectHelper.EstablishSslConnectionAsyncCore (System.IO.Stream stream, System.Net.Security.SslClientAuthenticationOptions sslOptions, System.Threading.CancellationToken cancellationToken) [0x00102] in <4807c99bd4cf4e79b711429bba2aeab1>:0
08:24:02:919	at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threadin
08:24:02:979	.CancellationToken cancellationToken) [0x00322] in <4807c99bd4cf4e79b711429bba2aeab1>:0
08:24:02:979	at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <4807c99bd4cf4e79b711429bba2aeab1>:0
08:24:02:979	at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boo
08:24:02:980	ean doRequestAuth, System.Threading.CancellationToken cancellationToken) [0x00089] in <4807c99bd4cf4e79b711429bba2aeab1>:0
08:24:02:980	at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <4807c99bd4cf4e79b711429bba2aeab1>:0
08:24:02:980	at System.Net.Http.HttpClient.FinishSendAsyncBuffered (Syste
08:24:03:020	.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <4807c99bd4cf4e79b711429bba2aeab1>:0

```

E-R-T-W, 2023-04-08:
Today, even scenario 4 (Release Compile and Deploy with no Debugger) crashed with 'out of memory' after only 6 calls to System.Net.Http.HttpClient.GetAsync():
```
09:01:36:709	Run...
09:01:36:990	pre HTTP requests
09:02:07:849	post HTTP request #1
09:02:19:834	post HTTP request #2
09:02:31:415	post HTTP request #3
09:02:42:563	post HTTP request #4
09:02:55:606	post HTTP request #5
09:03:07:224	post HTTP request #6
09:03:18:308	memalign returned null when requesting 1048576 bytes (1048576 al
09:03:18:308	ignment) -- out of memory?
09:03:32:690	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:03:32:690	code '142'" (142)
09:03:40:820	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:03:40:820	code '142'" (142)
09:04:03:704	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:04:03:704	code '142'" (142)
09:04:09:311	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:04:09:312	code '142'" (142)

```

E-R-T-W, 2023-04-08:
This may be the last post on the matter, just trying to throw you a bone and narrow down the problem further...
After running the code below with Release Compile and no Debugger, I get the following error after 27 calls to System.Net.Http.HttpClient.GetAsync()...

```
        public override Task Run()
        {
            Console.WriteLine("Run...");
            long i = 0;
            Task<HttpResponseMessage> responseTask;
            HttpResponseMessage response;
            System.Net.Http.HttpClient client;
            Console.WriteLine("pre HTTP requests");
            while (true)
            {
                client = new System.Net.Http.HttpClient();
                responseTask = client.GetAsync("https://weather.gc.ca/city/pages/ab-31_metric_e.html");
                responseTask.Wait();
                response = responseTask.Result;
                response = null;
                responseTask = null;
                client = null;
                i++;
                Console.WriteLine("post HTTP request #" + i);
            }
            return base.Run();
        }

```

Output window ends with:

```
09:41:43:158	post HTTP request #27
09:41:43:263	mono_fdhandle_insert: duplicate Socket fd 32
09:44:02:514	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:44:02:514	code '142'" (142)

```

E-R-T-W, 2023-04-10:
I've completed testing and research.
As it turns out, regardless of compilation to Release or Debug, and regardless of attaching the Debugger or not, the original code for this issue will eventually fail with 'out of memory', even if you set 
```
response = null;
```

However, I did find a true work around, although a hack! I know you aren't normally supposed to have to force Garbage Collection:
```
                GC.Collect();
                GC.WaitForPendingFinalizers();
```
but it completely and 100% solves the problem!

Someone more advanced than me will have to explain why forcing garbage Collection is required.
I tried explicitly disposing of both the response and client, as well as wrapping them in using clauses, to no avail.
I did a bunch of research and followed everything I could find on the web, but the ONLY thing that seems to works is to force Garbage Collection.

The code that works, vs the orignally posted code, is as follows:

```
    public class MeadowApp : App<F7FeatherV2>
    {
        private const string WIFI_NAME = "[SSID]";
        private const string WIFI_PASSWORD = "[Password]";
        private IWiFiNetworkAdapter wifi;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.Connect(WIFI_NAME, WIFI_PASSWORD, TimeSpan.FromSeconds(60)).Wait();
            return base.Initialize();
        }

        public override Task Run()
        {
            Console.WriteLine("Run...");
            long i = 0;
            Task<HttpResponseMessage> responseTask;
            HttpResponseMessage response;
            System.Net.Http.HttpClient client;
            Console.WriteLine("pre HTTP requests");
            while (true)
            {
                client = new System.Net.Http.HttpClient();
                responseTask = client.GetAsync("https://weather.gc.ca/city/pages/ab-31_metric_e.html");
                responseTask.Wait();
                response = responseTask.Result;
                GC.Collect();
                GC.WaitForPendingFinalizers();
                i++;
                Console.WriteLine("post HTTP request #" + i);
            }
            return base.Run();
        }
    }
```

Is this a true memory leak bug? Seems like it's a Garbage Collection bug to me (not running often enough).
If not a bug, can someone explain why forcing Garbage Collection is required?
Or can someone post what the corrrect alternative code should be to make forcing Garbage Collection unnecessary?

Thank you for your help!

zivkan, 2023-04-10:
HttpResponseMessage (the output of `HttpClient.GetAsync`) implements `IDisposable`.  Have you tried disposing it (for example, putting it in a `using` block)?

Theoretically, it shouldn't solve the problem, but sometimes these things are worth trying.

E-R-T-W, 2023-04-10:
Thanks for the reply, @zivkan! 
Yes tried both explicitly disposing, and also wrapping in using block. Both of those options have no affect - the meadow still crashes with an 'out of memory' error after repeated calls to System.Net.Http.HttpClient.GetAsync() i.e. it will work a few times, but eventually always run out of memory unless force Garbage Collections.

E-R-T-W, 2023-04-10:
It appears it's indeed a memory leak - **Even when forcing Garbage Collections, the Meadow eventually crashes with an 'out of memory error'** - it just takes significantly longer, 266 calls to System.Net.Http.HttpClient.GetAsync()
```
02:27:05:990	post HTTP request #265
02:27:11:007	memalign returned null when requesting 1048576 bytes (1048576 al
02:27:11:007	ignment) -- out of memory?
02:28:07:387	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
02:28:07:387	code '142'" (142)
```

Help! Thanks!

E-R-T-W, 2023-04-10:
Memory leak when forcing Garbage Collection is only when the debugger is attached.
I've successfully consecutively called System.Net.Http.HttpClient.GetAsync() over 1000 times while forcing Garbage Collection and it's still running with no 'out of memory' error when compiling to release and no debugger.

E-R-T-W, 2023-04-14:
@E-R-T-W Thanks for the help and support! You and the team and Wildeneress Labs have done an amazing job with the Meadow!

ladeak, 2023-04-24:
I experience a similar issue with IotHub connection (AMQP), after 2 requests it shows some sort of an error and crashes.

E-R-T-W, 2023-09-08:
Closing this issue - There's a work around by creating new instances of System.Net.Http.HttpClient() on the start of each loop, closing it and everything related to it at the end of the loop, and manually calling GC.Collect() and GC.WaitForPendingFinalizers()

ladeak, 2023-09-08:
Thank you, I will test it during the coming days

adrianstevens, 2023-10-31:
This one is definitely important

duduita, 2024-01-24:
After trying to reproduce this issue using v1.8.0.0 without a debugger attached, I couldn't reproduce scenario 2 (Debug Compile with No Debugger). After an overnight test, I could make more than 3000 requests, and keep going.

```
...
Meadow StdOut: post HTTP request #3277
Meadow StdOut: post HTTP request #3278
Meadow StdOut: post HTTP request #3279
Meadow StdOut: post HTTP request #3280
...
```

Could you check if you are still facing this issue on v1.8.0.0 in scenarios 2 and 4 (with No Debugger)? Please, remember to not only re-flash your device with the latest OS, but also re-deploy your app, since it'll update some needed .NET DLLs @E-R-T-W.

duduita, 2024-01-24:
After testing in three different devices, I confirmed that I'm not seeing this issue anymore on v1.8.0.0 in fact, when there's no debugger attached, since we should have a debugger leak, but there is another issue with it in specific. So, I'd appreciate if you could try to reproduce this issue after re-flashing and re-deploying your app using OS v1.8.0.0.

E-R-T-W, 2024-01-25:
> Never mind what I said above, I realized that I was not using 1.8.0.0, but a newer internal release. I'll double-check what I made that seemed to solve this leak.
>
> -

I can confirm that the original specific memory leak no longer exists, however, I have a new but very related issue that has taken me several days to track down just exactly where the bug lies. I don't know if it's a memory leak or not - no error - the ESP32 simply stops functioning i.e. with a continuous ping running, when the scenario happens, you can't even ping the ESP32. But there are no errors, no WiFi events triggered, the ESP32 simply goes away. Meadow still runs though as evidenced by the LED still blinking.

I'm almost ready to post source code along with instruction as to how to most easily replicate the issue, for WL Team to debug. You are getting very close, but this is still a significant bug.

Do you want a new issue? And close this one? Or would you like me to post to this one?

duduita, 2024-01-25:
> I can confirm that the original specific memory leak no longer exists, however, I have a new but very related issue that has taken me several days to track down just exactly where the bug lies. I don't know if it's a memory leak or not - no error - the ESP32 simply stops functioning i.e. with a continuous ping running, when the scenario happens, you can't even ping the ESP32. But there are no errors, no WiFi events triggered, and the ESP32 simply goes away. Meadow still runs though as evidenced by the LED still blinking.
> 
> I'm almost ready to post source code along with instruction as to how to most easily replicate the issue, for WL Team to debug. You are getting very close, but this is still a significant bug.
> 
> Do you want a new issue? And close this one? Or would you like me to post to this one?

Please correct me if I'm wrong @alexischr, but I think that it deserves a new issue @E-R-T-W, it sounds like an ESP32 issue. Then, let me know if you could confirm that this memory leak is not happening anymore on v1.8.0.0 to let us close this issue.

closed 2024-01-25 by duduita