WildernessLabs/Meadow_Issues #288: WiFi: iNetworkAdapter.IsConnected still reports true after iNetworkAdapter.Disconnect or WiFi goes down

E-R-T-W, 2023-04-12:
**Describe the bug**
Disconnect from WiFi AccessPointEither after by explicitly calling iNetworkAdapter.Disconnect() or reboot the AccessPiont, and iNetworkAdapter.IsConnected will still report 'true'. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run the following code which connects to WiFi then disconnects from WiFi. Once iNetworkAdapter.IsConnected == true after a successfull connection, it will never again report false, even after disconnecting (whether calling iNetworkAdapter.Disconnect() or rebooting the AccessPoint).

```
        private const string WIFI_NAME = "[SSID]";
        private const string WIFI_PASSWORD = "[Password]";
        private IWiFiNetworkAdapter wifi;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.NetworkConnected += Wifi_NetworkConnected;
            wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
            wifi.NetworkError += Wifi_NetworkError;
            return base.Initialize();
        }

        public override async Task Run()
        {
            Console.WriteLine("Run...");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == false

            try
            {
                // connect to the wifi network.
                Console.WriteLine($"Connecting to WiFi Network");
                await wifi.Connect(WIFI_NAME, WIFI_PASSWORD, TimeSpan.FromSeconds(60));
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Failed to Connect: {ex.Message}");
            }
            await Task.Delay(1000);

            while (!wifi.IsConnected) // this loop will not run since wifi.IsConnected == true as reported in the Wifi_NetworkConnected event
            {
                await Task.Delay(1000);
                Console.WriteLine("Waiting for wifi to connect loop...");
                Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // never executes
            };
            await Task.Delay(1000);

            await wifi.Disconnect(true);
            await Task.Delay(1000);
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == true even after wifi.Disconnect, even though wifi.Disconnect does truly disconnect WiFi

            try
            {
                // reconnect to the wifi network.
                Console.WriteLine($"Reconnecting to WiFi Network");
                await wifi.Connect(WIFI_NAME, WIFI_PASSWORD, TimeSpan.FromSeconds(60)); // attempting to reconnecting always times out
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Failed to reconnect: {ex.Message}");
            }

            while (!wifi.IsConnected)
            {
                await Task.Delay(1000);
                Console.WriteLine("Waiting for wifi to connect loop...");
                Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
            };
            await Task.Delay(1000);

            await wifi.ClearStoredAccessPointInformation();
            await Task.Delay(1000);
            Console.WriteLine("AccessPoint Info Cleared");

            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == true, still true even though WiFi is not connected and connot reconnect

            Console.WriteLine("Exiting...");
        }

        private void Wifi_NetworkError(INetworkAdapter sender, NetworkErrorEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkError event");
        }

        private void Wifi_NetworkDisconnected(INetworkAdapter sender)
        {
            Console.WriteLine("Wifi_NetworkDisconnected event");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
        }

        private void Wifi_NetworkConnected(INetworkAdapter sender, NetworkConnectionEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkConnected event");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == true
            wifi.ClearStoredAccessPointInformation().Wait();
            Console.WriteLine("AccessPoint Info Cleared");
        }

```

**Expected behavior**
iNetworkAdapter.IsConnected should report the correct status of the WiFi network.

**Screenshots**
The code is already decorated with the output results, but here's a copy of the Output:
`21:11:59:593	Initialize...
21:11:59:818	Run...
21:11:59:821	wifi.IsConnected = False
21:11:59:823	Connecting to WiFi Network
21:12:09:349	Wifi_NetworkConnected event
21:12:09:351	wifi.IsConnected = True
21:12:09:505	AccessPoint Info Cleared
21:12:10:479	Wifi_NetworkDisconnected event
21:12:10:479	wifi.IsConnected = True
21:12:11:518	wifi.IsConnected = True
21:12:11:520	Reconnecting to WiFi Network
21:13:12:902	Failed to reconnect: The operation has timed out.
21:13:14:967	AccessPoint Info Cleared
21:13:14:970	wifi.IsConnected = True
21:13:14:971	Exiting...
`

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 10 Pro v21H2
 - IDE and version: Visual Studio Professional 2022 v17.5.3
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.96.2

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
   - Meadow coprocessor version: 0.9.6.0 built 07 Mar 2023 21:05:31 UTC (c6708b58)

**Additional context**
Reconnction always fails too. Renders the WiFi network rather useless for any meaningful microController application.
I'll send a seperate bug report for the lack of WiFi reconnect issue.

E-R-T-W, 2023-04-14:
Easy and Reliable work around! The team at Meadow Labs have done an amazing job with the WiFi connectivity events:
           wifi.NetworkConnected += Wifi_NetworkConnected;
            wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
            wifi.NetworkError += Wifi_NetworkError;
The events can reliably be used to determine WiFi connectivity rather than relying on INetworkAdapter.IsConnected.

Thanks for all the continued hard work and efforts Wilderness Labs Team!

duduita, 2023-11-06:
I think that it was fixed in this [PR](https://github.com/WildernessLabs/Meadow.Core/pull/381/files), am I right, @halyssonJr?

halyssonJr, 2023-11-06:
Yes, you are right @duduita

closed 2024-03-26 by ctacke