WildernessLabs/Meadow_Issues #290: IWiFiNetworkAdapter.IPAddress not released after WiFi Disconnected

E-R-T-W, 2023-04-12:
**Describe the bug**
After a loss of WiFi connectivity, whether explicitly calling IWiFiNetworkAdapter.Disconnect() or by rebooting AccessPoint for example, IWiFiNetworkAdapter.IpAddress() still reports the DHCP supplied IP Address.
Due to the lack of the ability to Reconnect to WiFi, Issue #289, I'm trying to find a way to periodically check WiFi status/health, and Reset the Meadow in the event of WiFi connectivity loss. Unfortunately:
- IP Address is maintanied even after loss of WiFi connectivity - this particular Issue (I believe will be #290)
- Ping doesn't work - Issue #228 
- IWiFiNetworkAdapter.IsConnected always reports true - Issue #288 

I've exhausted all options I can think of to check for loss of connectivity, in order to Reset the Meadow.

Work around - but I'm not sure how reliable it is - the wifi.NetworkDisconnected event does seem to fire, and so one can execute Device.PlatformOS.Reset() in the event handler.

**To Reproduce**
Steps to reproduce the behavior:
1. Execute the following code and you will see that even after loss of WiFi connectivity, the IP Address is not released.

```
        private const string WIFI_NAME = "[SSID]";
        private const string WIFI_PASSWORD = "[Password]";
        private IWiFiNetworkAdapter wifi;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");
            wifi = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
            wifi.NetworkConnected += Wifi_NetworkConnected;
            wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
            wifi.NetworkError += Wifi_NetworkError;
            return base.Initialize();
        }

        public override async Task Run()
        {
            Console.WriteLine("Run...");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}"); // wifi.IsConnected == false
            Console.WriteLine("IPAddress = " + wifi.IpAddress);

            try
            {
                // Connect to the wifi network.
                Console.WriteLine($"Connecting to WiFi Network");
                await wifi.Connect(WIFI_NAME, WIFI_PASSWORD, TimeSpan.FromSeconds(60));
            }
            catch (Exception ex)
            {
                Console.WriteLine($"Failed to Connect: {ex.Message}");
            }
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");

            // Wait for both wifi connection and an IPAddress
            while (wifi.IsConnected == false || wifi.IpAddress.ToString() == "255.255.255.255")
            {
                Console.WriteLine("Waiting for wifi to connect loop");
                Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
                Console.WriteLine($"IPAddress = {wifi.IpAddress}");
                await Task.Delay(50);
            };
            Console.WriteLine("Waiting for wifi to connect loop ended");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
            Console.WriteLine($"IPAddress = {wifi.IpAddress}");

            await Task.Delay(1000);
            Console.WriteLine("Execute wifi.Disconnect now");
            await wifi.Disconnect(true);
            await Task.Delay(1000);
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
            Console.WriteLine($"IPAddress = {wifi.IpAddress}");

            // Would be nice to be able to wrap Device.PlatformOS.Reset() in some kind of diagnostic check to verify health of WiFi connection,
            // and Reset() as necessary, as a work around.
            // Until WiFi issues are resolved, however can't figure out a means to check status/health of WiFi becuase:
            // 1. Ping does not work - Issue #228.
            // 2. INetworkAdapter.IsConnected always reports true - Issue #288.
            // 3. INetworkAdapter.IpAddress keeps it's IP Address even after being disconnected - this current post Issue #290 (I believe it will be).
            // Note: IWiFiNetworkAdapter.NetworkDisconnected event does fire, but would still be good to run own WiFi diagnostics status check.
            //       IWiFiNetworkAdapter.NetworkDisconnected, along with IWiFiNetworkAdapter.NetworkError event, are currenlty the only work around
            //       I can find to execute Device.PlatformOS.Reset() to reboot to re-establish WiFi connectivity
            Console.WriteLine("Reseting now because because no other way to reconnect to network :(");
            Device.PlatformOS.Reset(); // In a production application, remove this line and let the IWiFiNetworkAdapter.NetworkDisconnected and NetworkError events Reset the Meadow
        }

        private void Wifi_NetworkError(INetworkAdapter sender, NetworkErrorEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkError event fired... -> Can execute Device.PlatformOS.Reset() here as only way to re-establish WiFi connectivity...");
            // Device.PlatformOS.Reset(); //uncomment this line for production applications
        }

        private void Wifi_NetworkDisconnected(INetworkAdapter sender)
        {
            Console.WriteLine("Wifi_NetworkDisconnected event fired... -> Can execute Device.PlatformOS.Reset() here as only way to re-establish WiFi connectivity...");
            Console.WriteLine($"wifi.IsConnected (from Wifi_NetworkDisconnected event) = {wifi.IsConnected}");
            // Device.PlatformOS.Reset(); //uncomment this line for production applications
        }

        private void Wifi_NetworkConnected(INetworkAdapter sender, NetworkConnectionEventArgs args)
        {
            Console.WriteLine("Wifi_NetworkConnected event fired...");
            Console.WriteLine($"wifi.IsConnected = {wifi.IsConnected}");
        }

```

**Expected behavior**
IWiFiNetworkAdapter.IPAddress should be released on loss of WiFi connectivity.

**Screenshots**
Output from the code is as follows:
```
16:33:58:921	Initialize...
16:33:59:130	Run...
16:33:59:134	wifi.IsConnected = False
16:33:59:239	IPAddress = 255.255.255.255
16:33:59:241	Connecting to WiFi Network
16:34:06:771	wifi.IsConnected = True
16:34:06:790	Waiting for wifi to connect loop
16:34:06:792	wifi.IsConnected = True
16:34:06:794	IPAddress = 255.255.255.255
16:34:06:851	Waiting for wifi to connect loop
16:34:06:853	wifi.IsConnected = True
16:34:06:855	IPAddress = 255.255.255.255
16:34:06:913	Waiting for wifi to connect loop
16:34:06:915	wifi.IsConnected = True
16:34:06:917	IPAddress = 255.255.255.255
16:34:07:419	Waiting for wifi to connect loop ended
16:34:07:421	wifi.IsConnected = True
16:34:07:424	IPAddress = 192.168.1.166
16:34:08:169	Wifi_NetworkConnected event fired...
16:34:08:171	wifi.IsConnected = True
16:34:08:430	Execute wifi.Disconnect now
16:34:08:577	Wifi_NetworkDisconnected event fired... -> Can execute Device.PlatformOS.Reset() here as only way to re-establish WiFi connectivity...
16:34:08:580	wifi.IsConnected (from Wifi_NetworkDisconnected event) = True
16:34:09:646	wifi.IsConnected = True
16:34:09:648	IPAddress = 192.168.1.166
16:34:09:649	Reseting now because because no other way to reconnect to network :(

```

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 10 Pro v21H2
 - IDE and version: Visual Studio Professional 2022 v17.5.4
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v0.96.2

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
   - Meadow coprocessor version: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC

**Additional context**
To diagnose/check WiFi health/status, I've also thought of trying to do something like GetWebPageViaHttpClient("https://postman-echo.com/get?foo1=bar1&foo2=bar2"), however the application will be behind a corporate firewall and can't reach out to the internet.
Is there any other way to check WiFi connectivity health/status (IP Address doesn't work Issue #290, Ping doesn't work Issue #228,  IsConnected doesn't work Issue #288).
Any suggestions?
Thanks!

E-R-T-W, 2023-04-14:
Easy and Reliable work around! The team at Meadow Labs have done an amazing job with the WiFi connectivity events:
wifi.NetworkConnected += Wifi_NetworkConnected;
wifi.NetworkDisconnected += Wifi_NetworkDisconnected;
wifi.NetworkError += Wifi_NetworkError;
The events can reliably be used to determine WiFi connectivity.
In the event of loss of WiFi connectivity, simply call Device.PlatformOS.Reset()

Wilderness Labs have a great product supported by a powerhouse team!
I'm enjoying the product(s) more and more, the more I use it.
Shout-out to the Wilderness Labs Team!
Well Done!

halyssonJr, 2024-04-04:
After disconnecting from AP, doesn't update the networking interface property (IP, SubnetMask, DnsAddress, and Gateway). This way we should refresh the interface info.

halyssonJr, 2024-04-15:
@E-R-T-W, the fix has been sent to Meadow. Core. Thanks for your report

ctacke, 2024-05-01:
Fixed in 1.11

closed 2024-05-01 by ctacke