WildernessLabs/Meadow_Issues #291: Unable to get data from an SSL end point

ctacke, 2023-04-14:
A simple web service with a public certificate is hosted at: [URL SENT TO SUPPORT]

Using a desktop browser and the API returns the current time with no issues.  However with a Meadow I get an error:

`GET FAILED: The SSL connection could not be established, see inner exception.(TLS initialization or handshake failed)`

Running on a Core Compute v2, Dual Ethernet (though same failure is reported by the customer on a Feather v2)
Firmware Versions
    OS: 0.9.7.23
    Mono: 0.9.7.23
    Coprocessor: 0.9.6.0
    Protocol: 7
    
Test code:
```
var client = new HttpClient();

var eth = Device.NetworkAdapters.First();
while (eth.IpAddress == IPAddress.None)
{
    Resolver.Log.Info($"IP: {eth.IpAddress}");
    await Task.Delay(1000);
}

Resolver.Log.Info($"IP: {eth.IpAddress}");

try
{
    var result = await client.GetAsync("https://empirestate.ddns.net:9444/api/DateTime");
    if (result.IsSuccessStatusCode)
    {
        var s = await result.Content.ReadAsStringAsync();
        Resolver.Log.Info($"GET RETURNED: {s}");
    }
    else
    {
        Resolver.Log.Info($"GET RETURNED: {result.StatusCode}");
    }
}
catch (Exception ex)
{
    if (ex.InnerException != null)
    {
        Resolver.Log.Info($"GET FAILED: {ex.Message}({ex.InnerException.Message})");
    }
    else
    {
        Resolver.Log.Info($"GET FAILED: {ex.Message}");
    }
}
```

DarkNoir, 2023-10-23:
I am having this issue on F7v2 and 1.3.4. I also tried using :
`HttpClient.ServerCertificateCustomValidationCallback = HttpClientHandler.DangerousAcceptAnyServerCertificateValidator` and my own validation callback delegate so I could validate the server cert as valid and that does not work either. This is a critical issue.

alexischr, 2024-05-14:
Meadow.OS has added support to make certificate validation optional since ver. 1.6.

closed 2024-05-14 by alexischr