WildernessLabs/Meadow_Issues #292: Wired ethernet does not get address when attached after boot

ctacke, 2023-04-17:
- Boot the device set up to use wired ethernet on a CCM but with no network cable plugged in.
- plug in the cable after the OS/App is running
- The network adapter never retrieves an address from the DHCP server.

If I boot with the cable plugged in, it successfully gets an address.

Environment:
CCM Dual ethernet board
```
Board Information
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 1D-00-1F-00-0D-50-4B-55-30-38-31-20
    Serial number: 20503857554B
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D5:7A:10

Firmware Versions
    OS: 0.9.7.22
    Mono: 0.9.7.22
    Coprocessor: 0.9.6.0
    Protocol: 7
```
Test Code:
```
var nic = Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();

nic.NetworkDisconnected += (s) =>
{
    Resolver.Log.Info("Network Disconnected");
};
nic.NetworkConnected += (s, e) =>
{
    Resolver.Log.Info($"Network Connected: {nic.IpAddress} | {e.IpAddress}");
};

while (true)
{
    Resolver.Log.Info($"{nic.IpAddress}");
    await Task.Delay(2500);
}
```

NevynUK, 2023-04-18:
It appears that the system attempts to start the ethernet interface and grab a DHCP address as the board starts.  If the system fails to get an ethernet address after three attempts then it fails the interface startup.

Peter-Moody, 2023-04-18:
Mark, your right, that's by design. Just a simplistic implementation.

patridge, 2023-04-18:
Out of curiosity, what would happen if it acquires an address successfully, but the connection is lost and reconnected? Does it do the same "3-5 times" thing before perma-failing? Or is it more resilient if it ever worked successfully?