WildernessLabs/Meadow_Issues #293: Creating a PWM port on D03 or D04 on Juego kills the display init

adrianstevens, 2023-04-20:
Juego v2.b had the piezos on D12 and D13

Juego v2.c moved then to D03 and D04 ... when updating the Juego core project, it will no longer initialize

```
Initialize...
Instantiating Jeugo v2 hardware
Initialize hardware...
Initializing I2C Bus
I2C initialized
Mcp23008 #1 initialized
Mcp23008 #2 initialized
Mcp23008 version initialized
Initializing SPI Bus
SPI initialized
App Error in App Initialize: Communication error.  Verify device is powered and that SCL is Connected.
Meadow.Hardware.NativeException: Communication error.  Verify device is powered and that SCL is Connected.
at Meadow.Hardware.I2cBus.DecipherI2CError (Meadow.Core.Interop+Nuttx+ErrorCode ec) [0x0003b] in <353af08b04ea4d0e945fcf4b8989fc0c>:0
at Meadow.Hardware.I2cBus.Write (System.Byte peripheralAddress, System.Span`1[T] data) [0x00091] in <353af08b04ea4d0e945fcf4b8989fc0c>:0
at Meadow.Hardware.I2cPeripheral.WriteRegister (System.Byte address, System.Byte value) [0x00057] in <de
```

ctacke, 2023-06-07:
This is an issue with the underlying OS driver for PWM.  PWMs will step on the pin configuration for anything else in the same processor timer group.  What this means is that, for example, if you create an I2C bus, which has pins on TIM4, then create a PWM on D03, which is also TIM4, it will reconfigure the I2C pins and the bus will stop working.

For now, the workaround is to always create your PWMs first in your app configuration.  The OS team is investigating fixing the underlying driver.

adrianstevens, 2023-11-07:
Also hitting this issue with the Mikrobus pins on Project Lab v3

adrianstevens, 2024-02-20:
and hitting it again on the GNSS tracker

Peter-Moody, 2024-02-22:
Found that whenever a PWM was configured, within meadow-upd.c, all 4 channels were also being configured as PWM output channels. @ctacke is looking at a possible solution from the managed side.

adrianstevens, 2025-02-04:
I'll re-test

adrianstevens, 2026-02-28:
looks like it fixed - closing issue