WildernessLabs/Meadow_Issues #295: Meadow in low-power mode still consumes about 30ma

Peter-Moody, 2023-05-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [e.g., Windows 10 v22H2, macOS Monterey v12.6]
 - IDE and version: [e.g., Visual Studio Professional 2022 v17.3.5, Visual Studio 2022 for Mac v17.3.5]
 - Meadow extension for IDE version: [e.g., VS 2022 Tools for Meadow v0.19.9]

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: [e.g. F7v2, CCMv2]
 - Meadow OS version: [e.g., v0.6.7.19]
 - If they are different, please provide these versions as well.
   - Meadow Mono version: [e.g., v0.6.7.19]
   - Meadow coprocessor version: [e.g., v0.6.7.19]

**Additional context**
Add any other context about the problem here. If you are working with additional external hardware/sensors, please list them here.

Peter-Moody, 2023-05-01:
Found this comment in the F7's data sheet pp 84.

1. NC (not-connected) pins are not bonded. They must be configured by software to output push-pull and forced to 0 in the output data register to avoid an extra current consumption in low-power modes. list of pins: PI8, PI12, PI13, PI14, PF6, PF7, PF8, PF9, PC2, PC3, PC4, PC5, PI15, PJ0, PJ1, PJ2, PJ3, PJ4, PJ5, PH6, PH7, PJ12, PJ13, PJ14, PJ15, PG14, PK3, PK4, PK5, PK6 and PK7.

This is currently not done.

alexischr, 2023-05-02:
@Peter-Moody can we find out if that is mostly the ESP32 using the power? If so, at least this is an issue that that ESP sleep will resolve.

Peter-Moody, 2023-11-22:
Note for implementation. In STM32F777 Ref Manual see top pp 137

> The ADC or DAC can also consume power during the Stop mode, unless they are disabled
> before entering it. To disable them, the ADON bit in the ADC_CR2 register and the ENx bit
> in the DAC_CR register must both be written to 0.