WildernessLabs/Meadow_Issues #296: Need ability to control linking/trimming

ctacke, 2023-05-06:
Through Beta 3.1 the linker behavior is opaque and troublesome.  A super-common problem is a failure in Maple request handlers, since they are loaded via reflection.  The linker is removing handlers, or parts of POCOs that the handlers use.

Right now we have the ability to skip linking on a per-assembly basis, but it's only available in the CLI:

```
> $ meadow app deploy --nolink MyAssembly -f App.dll
```

We need a better, and preferrably more granular mechanism to control trimming that integrates with the tools.

adrianstevens, 2023-05-06:
or we use attributes that are read at build time

johnkattenhorn, 2023-07-12:
@ctacke - should this issue to annotated with the upcoming work you and I tested ?

ctacke, 2023-07-14:
Yes, it needs documentation, but is otherwise ready

lamebrain, 2023-09-05:
@patridge did you work on documentation for this?

patridge, 2023-09-05:
> @patridge did you work on documentation for this?

Not sure. I created some docs on various Meadow config files, but not necessarily the command line parameters.
http://developer.wildernesslabs.co/Meadow/Meadow.OS/Configuration/

closed 2024-03-26 by ctacke