WildernessLabs/Meadow_Issues #297: AMQP SenderLink times out on 3 call

ladeak, 2023-05-06:
**Describe the bug**

The third time I send a message using Ampq SenderLink, I get timeout exception.

**To Reproduce**

```csharp
using System;
using System.Buffers;
using System.Globalization;
using System.IO;
using System.Net;
using System.Security.Cryptography;
using System.Text;
using System.Text.Json;
using System.Threading.Tasks;
using Amqp;
using IotProcessor.Data;
using Meadow;

public class IotHubService
{
    private const string HubName = "myIotHubName";

    internal readonly string DeviceId = Resolver.Device.Information.UniqueID;

    private SenderLink _sender;
    private ReceiverLink _receiver;

    public async Task StartAsync()
    {
        try
        {
            Connection.DisableServerCertValidation = false;
            // parse Azure IoT Hub Map settings to AMQP protocol settings
            // https://learn.microsoft.com/en-us/azure/iot-hub/iot-hub-dev-guide-sas?tabs=node
            var hostName = HubName + ".azure-devices.net";
            var userName = DeviceId + "@sas." + HubName;
            var token = GenerateSasToken($"{hostName}/devices/{DeviceId}", Secrets.DevicePrimaryKey, string.Empty);

            var factory = new ConnectionFactory();
            var connection = await factory.CreateAsync(new Address(hostName, 5671, userName, token));
            var session = new Session(connection);
            _sender = new SenderLink(session, "send-link", $"devices/{DeviceId}/messages/events");
            _receiver = new ReceiverLink(session, "receive-link", $"devices/{DeviceId}/messages/deviceBound");
            _receiver.Start(1, OnMessage);
            _receiver.SetCredit(1, true);
        }
        catch (Exception ex)
        {
            Resolver.Log.Error(ex.Message);
        }
    }

    public async Task StopAsync()
    {
        await _sender.CloseAsync();
        await _receiver.CloseAsync();
    }

    private void OnMessage(IReceiverLink receiver, Message message)
    {
        try
        {
        }
        finally
        {
            receiver.Accept(message);
        }
    }

    public void Send(string data)
    {
        var message = new Message(data);
        _sender.Send(message);
    }

    private string GenerateSasToken(string resourceUri, string key, string policyName, int expiryInSeconds = 3600 * 8)
    {
        var fromEpochStart = DateTime.UtcNow - new DateTime(1970, 1, 1);
        var expiry = Convert.ToString((int)fromEpochStart.TotalSeconds + expiryInSeconds);

        var stringToSign = WebUtility.UrlEncode(resourceUri) + "\n" + expiry;

        var hmac = new HMACSHA256(Convert.FromBase64String(key));
        var signature = Convert.ToBase64String(hmac.ComputeHash(Encoding.UTF8.GetBytes(stringToSign)));

        var token = string.Format(CultureInfo.InvariantCulture, "SharedAccessSignature sr={0}&sig={1}&se={2}", WebUtility.UrlEncode(resourceUri), WebUtility.UrlEncode(signature), expiry);

        if (!string.IsNullOrEmpty(policyName))
        {
            token += "&skn=" + policyName;
        }

        return token;
    }
}
```

Call `Send` method 3 times.

**Expected behavior**

Should behave as the first two case: no timeouts

**Meadow.Foundation (please complete the following information as best as you can):**

    <PackageReference Include="Meadow.F7" Version="0.*" />
    <PackageReference Include="Meadow.Foundation" Version="0.*" />
    <PackageReference Include="Meadow.ProjectLab" Version="0.*" />
    <PackageReference Include="Meadow.AmqpNetLite" Version="*" />

(currently Meadow.AmqpNetLite 0.96.0)

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: VS Code (latest)
 - Meadow 0.96.2

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow  Hardware version: F7FeatherV2
 - Meadow OS version: OS: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
 - Meadow Mono version: Mono: 0.9.6.3 built 30 Mar 2023 00:26:46 UTC
 - Meadow coprocessor version: Coprocessor: 0.9.6.0 built 07 Mar 2023 21:05:31 UTC (c6708b58)

I have a projectlab attached.

halyssonJr, 2025-01-30:
@ladeak, I had a chance to test using the public OS version (2.0.0.3) and also the internal release 2.0.5.0, and works perfectly. Let me share some results. I made +100 requests to my Azure IoT hub, and I didn't receive any timeout message.

- Meadow CLI output: 
```
[application] Sending a message ...
[application] Data Sent : {"counter":98}
[application] Sending a message ...
[application] Data Sent : {"counter":99}
```
- Screenshot from Azure Website

[image]
- Azure App
[Azure_Issue.meadow-project-lab.zip](https://github.com/user-attachments/files/18606425/Azure_Issue.meadow-project-lab.zip)

ladeak, 2025-01-30:
Thank you, I have not tested it lately, without disposing. Thank you for investigating.

closed 2025-01-30 by ladeak