WildernessLabs/Meadow_Issues #300: Multiple network configurations for wifi.config.json

ladeak, 2023-05-08:
**Is your feature request related to a problem? Please describe.**

I would like to use the device across multiple wifi networks: ie. in the office connecting to office WIFI, at home connecting home WIFI.

**Describe the solution you'd like**
I would like to configure multiple networks in wifi.config.json. Ideally on device startup it could find the first available network and connect to it.

**Describe alternatives you've considered**
Considered doing this in code by manually connecting the networks.

NevynUK, 2023-09-18:
This functionality could be implemented in the application itself by combining scan for access points to determine the availability of a particular access point and then use credentials stored with the application.

closed 2023-09-18 by NevynUK

ladeak, 2023-09-18:
I think this was asked on a support chat on Slack to create a backlog item for this. The workaround is known, but this was discussed as a convenient feature.