WildernessLabs/Meadow_Issues #301: NetworkConnected event not fired correctly when AutomaticallyStartNetwork is configured

NevynUK, 2023-05-17:
**Describe the bug**
NetworkConnected event not fired correctly when AutomaticallyStartNetwork is configured.

**To Reproduce**

Create a `meadow.config.yaml` file with the following contents:

```yaml
Network:
    Interfaces:
        - Name: WiFi
          UseDHCP: true
          IPAddress: 192.168.1.10
          NetMask: 255.255.255.0
          Gateway: 192.168.1.254
    DefaultInterface: WiFi
Coprocessor:
    AutomaticallyStartNetwork: true
```

Now for the C# application, add the following code to the application (WiFi Basics can be used as a starting point):

```c#
INetworkAdapter adapter;

if (Device.PlatformOS.SelectedNetwork == IPlatformOS.NetworkConnectionType.Ethernet)
{
    Resolver.Log.Info("Configured for wired ethernet.");
    adapter = Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();
}
else
{
    Resolver.Log.Info("Configured for WiFi.");
    adapter = Device.NetworkAdapters.Primary<IWiFiNetworkAdapter>();
    IWiFiNetworkAdapter wifi = (IWiFiNetworkAdapter) adapter;
    Resolver.Log.Info($"Automatically connect to network: {wifi.AutoConnect}");
    Resolver.Log.Info($"Default access point: {wifi.DefaultSsid}");
}

adapter.NetworkConnected += Adapter_NetworkConnected;

private void Adapter_NetworkConnected(INetworkAdapter sender, NetworkConnectionEventArgs args)
{
    Resolver.Log.Info($"Connected to network");
}
```

The connection event does not get raised to the managed code.

NevynUK, 2023-06-18:
Tested using 1.0.3.0 and the events are being raised from the ESP through to the STM.

Generated a test application and asked for homework to be marked.

NevynUK, 2023-06-19:
Interesting, the following code works:

```c#
wifi.NetworkConnected += Wifi_NetworkConnected;
while (!wifi.IsConnected)
{
    Thread.Sleep(1000);
}
```

But this code does not:

```c#
wifi.NetworkConnected += Wifi_NetworkConnected;
while (!wifi.IsConnected);
```

Reducing the `Thread.Sleep` value down to 1ms still works but the absence of the `Thread.Sleep` causes the system to appear to hang even though communication with the board is still possible.

NevynUK, 2023-06-19:
I have tried increasing the thread priority for the two threads dealing with the ESP and this has no effect.

bryancostanich, 2023-10-17:
Likely caused by #311.

halyssonJr, 2023-10-19:
I tested using with WiFi and NetworkConnected event  works.
 ```
    Board: F7FeatherV2
    OS: 1.3.5.6
    Mono: 1.3.5.6
    Coprocessor: 1.3.4.0
```
Output: 
```
Meadow StdOut: Initializing OS...
Meadow StdOut: Parsing app.config.yaml...
Meadow StdOut: Log level: Information
Meadow StdOut: Device is configured to use WiFi for the network interface
Meadow StdOut: Update Service is disabled.
Meadow StdOut: Health Metrics disabled.
Meadow StdOut: WiFi adapter not connected.
Meadow StdOut: Try connecting ...
Meadow StdOut: Try connecting ...
Meadow StdOut: Try connecting ...
Meadow StdOut: Try connecting ...
Meadow StdOut: Try connecting ...
Meadow StdOut: Connected :) | 0
Meadow StdOut: Connected :) | 1
Meadow StdOut: WiFi connected.
Meadow StdOut: Connected :) | 2
Meadow StdOut: Connected :) | 3
Meadow StdOut: Connected :) | 4
...
Meadow StdOut: Connected :) | 19
Meadow StdOut: Connected :) | 20
Meadow StdOut: WiFi disconnected (Event .
Meadow StdOut: Connected :) | 0
...
Meadow StdOut: Connected :) | 7
Meadow StdOut: Connected :) | 8
Meadow StdOut: Connected :) | 9
```
Below the C# code to test
[Test.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13046969/Test.zip)

duduita, 2023-10-30:
The network events seem working as expected on WiFi. I realized that the `IsConnect` property was not being updated properly, but after using the latest Meadow.Core `develop` branch, which includes the fix [#381](https://github.com/WildernessLabs/Meadow.Core/pull/381), it worked well.

For Ethernet, the network events are not working, so I created an issue on Meadow.Core for this implementation:
- https://github.com/WildernessLabs/Meadow.Core/issues/388

closed 2023-10-31 by bryancostanich