WildernessLabs/Meadow_Issues #305: Meadow fails to deploy correctly due to post build chain (linking?) when the Meadow Application is named a certain way

asalvo, 2023-05-21:
**Describe the bug**
Given a Meadow application named `Meadow.Hw.App.FeatherV1` when you deploy the application, Meadow.dll (along with other dlls) are missing from the prelink_bin folder and there are no files in the postlink_bin folder. Only the files in the prelink_bin file get deployed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new Meadow project with a solution named MeadowApplication and a project name of Meadow.hw.App.FeatherV1

[image]

2. I have a FeatherV1 board, so I changed the App<T> in MeadowApp.cs to FeatherV1

3. Right click on the application in Visual Studio 2022 and click deploy 

**Expected behavior**
The application should be built and deploy successfully regardless of the project name. If a project name is not valid for some reason then a compile time error should be raised. If that's not possible, then a error from the Meadow CLI should be clearly displayed. I also think the Meadow CLI should clearly show an error if the postlink_bin folder is empty as these seems like an error state that should never occur.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 22H2 (22621.1555)
 - IDE and version: Visual Studio 2022 17.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow (Preview) v1.0.0
 - meadow --version: v1.0.0

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV1
- OS: 1.0.0.0
- Mono: 1.0.0.0
- Coprocessor: 1.0.0.0
- Protocol: 7

**Additional context**
It could be the name of the project, the namespace, or the folder name of the project. Once I was able to get a simple repro I gave up trying to figure out why my original project wouldn't work. 

Meadow.App.FeatherV1 does not appear to work either. 
[image]

adrianstevens, 2024-03-26:
This is fixed in the CLI v2.x code ... we need to update the extensions

ctacke, 2024-09-17:
@adrianstevens did this get fixed that you know?

alexischr, 2024-12-10:
This is fixed, @ctacke fixed this.

closed 2024-12-10 by alexischr