WildernessLabs/Meadow_Issues #311: Thread scheduler is not doing round-robin

ctacke, 2023-05-31:
If you have a thread in an application that is doing a tight-loop, other threads are never serviced.  Adding a `Thread.Yield()` allows them to be serviced.

Also, raising the priority of the other thread has no effect (i.e. it's still never serviced)

OS and Core v 1.0.0

Expected behavior: Even with a tight loop, threads should get serviced in a round-robin fashion.

Repro:

Using an F7v2, loop back a pair of pins with a jumper.
Set a PinPair to match your loopback:
https://github.com/WildernessLabs/Meadow.TestSuite/blob/a37e7a2dfefe38b68ef5c9ee99f602e4b6cf69f6/src/Validation/loopback/DiscreteLoopback/MeadowApp.cs#L23

And run any of these Counter tests:
https://github.com/WildernessLabs/Meadow.TestSuite/blob/a37e7a2dfefe38b68ef5c9ee99f602e4b6cf69f6/src/Validation/loopback/DiscreteLoopback/MeadowApp.cs#L65

The test will, right now, succeed.

Remove either or both of these Yield calls (that should not be necessary) to see it fail:
https://github.com/WildernessLabs/Meadow.TestSuite/blob/a37e7a2dfefe38b68ef5c9ee99f602e4b6cf69f6/src/Validation/loopback/DiscreteLoopback/CounterBase.cs#L58
or
https://github.com/WildernessLabs/Meadow.TestSuite/blob/a37e7a2dfefe38b68ef5c9ee99f602e4b6cf69f6/src/Validation/loopback/DiscreteLoopback/CounterBase.cs#L89

ctacke, 2023-07-24:
Tested with new internal OS release setting up round-robin in Nuttx, and the issue persists.  I think we need to change something in Mono.  Passed over to OS team for investigation.

doingnz, 2023-10-02:
Please also consider https://github.com/WildernessLabs/Meadow.Foundation/issues/745
as it gives another way to stress the Task Schedular using regular Blinky Demo application.  

Add 3 x tasks that compete to set the LED color.  Adding Thread.Yield() or Task.Delay() helps, but does not provide an effective work-around as App is incorrectly shutdown and only 2 of 3 tasks continue to be scheduled. 

```
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(10);
                    Resolver.Log.Info("Red LED");
                    await ShowColorPulse(Color.Red, defaultDuration / 2);
                    Thread.Yield();
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(10);
                    Resolver.Log.Info("Blue LED --------------------");
                    await ShowColorPulse(Color.Blue, defaultDuration / 2);
                    Thread.Yield();
                }
            });
            Task.Run(async () =>
            {
                while (true)
                {
                    //await Task.Delay(10);
                    Resolver.Log.Info("Green LED");
                    await ShowColorPulse(Color.Green, defaultDuration / 2);
                    Thread.Yield();
                }
            });
```
The first challenge is this code will very quickly shutdown the application with the following error
```
App Error in App Run: A task was canceled.
System.Threading.Tasks.TaskCanceledException: A task was canceled.
at Meadow.Foundation.Leds.RgbPwmLed.StopAnimation () <0xc13623d8 + 0x000d8> in <d663bcbce379408896530e841a6940d0>:0
at Meadow.Foundation.Leds.RgbPwmLed.StartPulse (Meadow.Foundation.Color color, System.TimeSpan pulseDuration, System.Single highBrightness, System.Single lowBrightness) <0xc1361d18 + 0x000d4> in <d663bcbce379408896530e841a6940d0>:0
at MeadowBlinkyTest.MeadowA
p.ShowColorPulse (Meadow.Foundation.Color color, System.TimeSpan duration) [0x00054] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:125
at MeadowBlinkyTest.MeadowApp.CycleColors (System.TimeSpan duration) [0x00051] in C:\Users\centr\source\repos\MeadowBlinkyTest\MeadowApp.cs:116
at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) <0xc0bf2840 + 0x00
18> in <d4f614572c6047c08610bc5349864be2>:0
App shutting down
Shutdown
```
If you add the suggested change work-a-round to Meadow.Foundation to handle the unexpected TaskCancelledException, then the code will run for longer before the main task ends. The 2 x Tasks competing will continue execute, but the third appears to stop being scheduled. 

These bugs are still present in 1.3.4.0

bryancostanich, 2023-10-31:
Alexis has a partial fix for this by registering all the mono threads into the round robin scheduler so they can participate. Lots of artifacts from the fix, so this will be a bigger lift.

bryancostanich, 2023-11-03:
We are now testing a build internally with, hopefully, a complete fix for this issue. All mono threads should be participating in the scheduler as appropriate. Will keep the issue updated with our findings.

bryancostanich, 2023-11-06:
Fixed, on @ctacke's plate to validate.

jorgedevs, 2023-11-06:
Testing:
- On a F7 Feather wired with Discreet Loop back
- Board is running Meadow.OS with RR (Internal Build)
- Running the `DiscreetLoopback_v2` application
- Running the app as-is: test passes

I reduced the test to only `new CounterRisingAB(),`

On `CounterBase` class, if I comment out the `Thread.Sleep(1);` in line 58, im getting: 

```
Initializing OS...
Parsing app.config.yaml...
Log level: Debug
GC thread 15 is now round-robin
Device is configured to use WiFi for the network interface
Update Service is disabled.
Health Metrics disabled.
Starting validation tests...
1000 pulses took 107ms
Running CounterRisingAB...
GC thread 16 is now round-robin
PIN A00 expected 50 events but received 51
PIN D02 does not support interrupts. Skipping.
PIN D14 does not support interrupts. Skipping.
Tests complete.
---- FAILED TESTS----
CounterRisingAB
GC thread 17 is now round-robin
GC thread 18 is now round-robin
GC thread 19 is now round-robin
GC thread 20 is now round-robin
```

On `CounterBase` class, if I comment out the `Thread.Sleep(1);` in line 89, im getting: 

```
Initializing OS...
Parsing app.config.yaml...
Log level: Debug
GC thread 15 is now round-robin
Device is configured to use WiFi for the network interface
Update Service is disabled.
Health Metrics disabled.
Starting validation tests...
1000 pulses took 112ms
Running CounterRisingAB...
GC thread 16 is now round-robin
PIN A00 expected 50 events but received 0
PIN A02 expected 50 events but received 0
PIN A04 expected 50 events but received 0
PIN SCK expected 50 events but received 0
PIN CIPO expected 50 events but received 0
PIN D02 does not support interrupts. Skipping.
PIN D03 expected 50 events but received 0
PIN D05 expected 50 events but received 0
PIN D07 expected 50 events but received 0
PIN D09 expected 50 events but received 0
PIN D11 expected 50 events but received 0
PIN D14 does not support interrupts. Skipping.
Tests complete.
---- FAILED TESTS----
CounterRisingAB
GC thread 17 is now round-robin
GC thread 18 is now round-robin
GC thread 19 is now round-robin
GC thread 20 is now round-robin
```

ctacke, 2023-11-08:
More testing with instrumented Core, Mono and OS code indicates that the place the interrupt stack hangs is actually in the IST call to the interrupt handler event's `Invoke()` command.  Two action items from this discovery:
1. @alexischr and @ctacke to think about wtf this would cause it to pause
2. @ctacke to investigate possible more direct route to event handlers that do not use `Invoke()`

doingnz, 2023-11-09:
I am sure you are aware, but as I have been bitten by this a few times in the past, a gentle reminder that adding instrumentation can easily change the dynamics of code flow and produce unexpected results or actually break the code. e.g. if the instrumentation is text output and requires access to shared hardware or I/O semaphores, adding it can create new race conditions not present in production software. The slow speed of text instrumentation can significantly impact how the code runs when it's added to fast interrupts. I find it helpful to use a GPIO port to flag which bit of code is running and then trace with a logic analyser (I am a happy customer of Salea Logic). e.g. set a bit at the start of an ISR and clear when the ISR completes and normal execution continues. 

Another trap I have fallen into is nested interrupts where one interrupt is not serviced as the CPU is still in the context of another earlier interrupt. In some CPU, it is necessary to clear a flag at the beginning of an ISR to permit other higher priority interrupts to interrupt the current interrupt.

ctacke, 2023-11-09:
Testing of the 1.5.0.3 binaries shows that round-robin is now actually occurring.  The IST is running at the same priority as all other managed threads, so they are not truly "interrupting" but they are at least getting serviced now so this bug is fixed.  We'll create another issue to discuss how to proceed with possible IST priority changes or capabilities

doingnz, 2023-11-09:
when do we get to play with this goodness? :-)

doingnz, 2023-11-10:
@ctacke 1.5.0.3 is looking good!!!

I added notes and trace under [Meadow.Foundation-#745](https://github.com/WildernessLabs/Meadow.Foundation/issues/745) that stress tests 4 tasks fighting to set the RGB LED.

bryancostanich, 2023-11-30:
Not _quite_ in the clear yet. Still some lingering edge-cases we're working through. Targeted for release in v1.6 now.

bryancostanich, 2023-11-30:
For more context, the fix is in, but it has uncovered an underlying bug in Mono, in how it interacts with the scheduler and what priorities are set. @alexischr is working through it.

alexischr, 2024-01-04:
Our round-robin effort continues, and we are close, with one remaining bug when using UART but an an otherwise seemingly fully-working Meadow that passes all other tests. Many interruptible POSIX calls had now had to have their interrupt error code (`EINTR`) handled, and it's possible we are missing some more.

bryancostanich, 2024-01-09:
@adrianstevens @ctacke; @alexischr needs help repro'ing this, he's not seeing it anymore but he's also not seeing it on old bins, so either something got fixed in managedlandia, or Alexis just needs a second set of eyes to make sure he's doing it right.

bryancostanich, 2024-02-15:
Woohoo! Update, this is fixed, working and being tested internally!

johnkattenhorn, 2024-02-15:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Whoop, Whoop!

closed 2024-02-27 by bryancostanich