WildernessLabs/Meadow_Issues #315: Meadow.Cloud - New Collection Failed

jeffjulian, 2023-06-05:
**Describe the bug**
Meadow.Cloud failed when I tried to create a Collections called, "Abraham"

**To Reproduce**
Steps to reproduce the behavior:

1. Login for the first time with my already established Wilderness Labs account
2. Click Collections
3. Click New Collection
4. Enter name "Abraham" in the Collection Name field
5. Click add Create button
6. Error - {"type":"[https://tools.ietf.org/html/rfc7231#section-6.5.1","title":"One](https://tools.ietf.org/html/rfc7231#section-6.5.1%22,%22title%22:%22One) or more validation errors occurred.","status":400,"traceId":"00-858b2266ede04bc59c9aae7946a6c093-243da6396ad3541d-01","errors":{"OrgId":["The OrgId field is required."]}}

**Expected behavior**
New Collection should have been added

**Screenshots**
[image]

lamebrain, 2023-06-05:
Repro: 
Sign out and sign back in (prod only). Page won't load.

Console error:
[image]

adunndevster2, 2023-06-05:
Strange... it's doing this odd thing where the base url is innacurate...
[image]

adunndevster2, 2023-06-05:
@lamebrain - this should fix it: https://github.com/WildernessLabs/Meadow.Cloud.UI/pull/58

🤦‍♂️

lamebrain, 2023-06-06:
Yup, deployed and verified. Thanks!

closed 2023-06-06 by lamebrain