WildernessLabs/Meadow_Issues #320: Add HCOM support for "current directory"

ctacke, 2023-06-12:
Currently HCOM can only work with a single folder: `/meadow0`.  So any request to write, read or list files is locked to this folder.

To allow reading, writing and listing files, I propose we add the concept of a "current directory" to HCOM, and it would initialize to "/meadow0".  This would maintain compatibilty with all existing functions.

However if we did a "set working directory" to "/meadow0/data" we could then read, write and list files inside that child directory.

We should also add a "list folders" method to allow directory traversal.

So the commands we need are:
- GET_CURRENT_FOLDER
- SET_CURRENT_FOLDER
- LIST_FOLDERS

Peter-Moody, 2023-06-12:
Additional information
only subdirs of /meadow0?  Yes, that's totally fair
auto-create folders? Yes, that simplifies things
delete dir?  For now, we don't.  That's easy!

Peter-Moody, 2023-10-19:
While  looking at the issue, also check on difficulty of adding SD Card support for read/write.

Peter-Moody, 2023-11-04:
Input from Chris Tacke 4Nov23 related to the FIle structure returned  when I do a list files?

`meadow list files` : returns the list of files and folders in the root (/meadow0)
`meadow list files {path} `: returns the list of files and folders in the specified folder
When returning the list, if it's a file, return the file CRC, if it's a folder, just set the CRC to 0xffffffff

Peter's reply
`meadow list files` : returns the list of files and folders in the root (/meadow0)
On each line, one file *or* directory. For a file, no proceeding directory information, just the file name.
`meadow list files {path} `:  like the above, but instead of `/meadow0` it uses the {`path`} for returning file/directory list.
Beside CRC, could you like any file stats (e.g.  file size)?
As I understand this, someone needing to find a file by name, might need to do a lot of searching. Conclusion. this is intended for `Meadow.Core` automated use only.

closed 2024-03-26 by ctacke