WildernessLabs/Meadow_Issues #323: Scan for access points sometimes returns corrupt information

NevynUK, 2023-06-19:
Scanning for network access points sometimes shows what looks like binary data in the output:

```
Meadow StdOut: |-------------------------------------------------------------|---------|
Meadow StdOut: |         Network Name             | RSSI |       BSSID       | Channel |
Meadow StdOut: |-------------------------------------------------------------|---------|
Meadow StdOut: | BTWi-fi                          |  -70 |      7AB687C5BD74 |    11   |
Meadow StdOut: | MarkS                            |  -71 |      ACB687C5BD73 |    11   |
Meadow StdOut: |                                  |  -72 |      7AB687C5BD77 |    11   |
Meadow StdOut: | Guest Wifi                       |  -84 |      E2516316487C |     6   |
Meadow StdOut: |                                  |  -89 |      8C85807A497A |     1   |
Meadow StdOut: | BT-MRF6CW                        |  -91 |      4C22F37B9D82 |    11   |
Meadow StdOut: | York                             |  -92 |      04D4C41C8809 |     1   |
Meadow StdOut: | EE-HQR6W9                        |  -92 |      F04DD4487CBA |    11   |
Meadow StdOut: |                                  |  -92 |      EA4DD4487CBE |    11   |
Meadow StdOut: | BTWi-fi                          |  -92 |      7222F37B9D83 |    11   |
Meadow StdOut: | SKY4BRNL                         |  -93 |      3C8994094DDA |     1   |
Meadow StdOut: | FRITZ!Box 7530 GB                |  -93 |      DC396FF74E7D |     6   |
Meadow StdOut: | Burnside                         |  -94 |      B0E4D5838966 |     6   |
Meadow StdOut: | BurnsideGuest                    |  -94 |      B2E4D5838964 |     6   |
Meadow StdOut: | BT-SRA597                        |  -95 |      4C1B86B59349 |    11   |
Meadow StdOut: |                                  |  -96 |      6A1B86B5934D |    11   |
Meadow StdOut: | /                                |   95 |      42435E006232 |    48   |
Meadow StdOut: | Ñ#0¥Ëm? Û                      |   33 |      C3E1CEE17A7F |   240   |
```

Method:

```c#
async Task ScanForAccessPoints(IWiFiNetworkAdapter wifi)
{
    Console.WriteLine("Getting list of access points.");
    var networks = await wifi.Scan(TimeSpan.FromSeconds(60));

    if (networks.Count > 0)
    {
        Console.WriteLine("|-------------------------------------------------------------|---------|");
        Console.WriteLine("|         Network Name             | RSSI |       BSSID       | Channel |");
        Console.WriteLine("|-------------------------------------------------------------|---------|");

        foreach (WifiNetwork accessPoint in networks)
        {
            Console.WriteLine($"| {accessPoint.Ssid,-32} | {accessPoint.SignalDbStrength,4} | {accessPoint.Bssid,17} |   {accessPoint.ChannelCenterFrequency,3}   |");
        }
    }
    else
    {
        Console.WriteLine($"No access points detected.");
    }
}
```

NevynUK, 2023-10-26:
The uplift of the IDF to 5.1 has introduced an error where scanning for access points is causing the ESP32 to reboot and the errors seem to suggest heap corruption.

bryancostanich, 2023-10-31:
need to get the uplift out of `main` branch so we can ship `v1.5`.  Also, the uplift is lower priority.

bryancostanich, 2023-10-31:
Also, dropping the priority on this, given that it's an artifact of the uplift.