WildernessLabs/Meadow_Issues #324: `meadow file initial -f file_name` appears to be broken

NevynUK, 2023-06-19:
**Describe the bug**
`meadow file initial -f meadow.config.yaml` returns the following:

```
Connecting to Meadow on /dev/tty.usbmodem3360317434381

Getting files on partition 0
File '/meadow0/' cannot be opened
Read 0 bytes from meadow.config.yaml:
Done!
```

I have traced this through to the method in the OS that should put together the response and the file name is an empty string which would explain the output above.

CartBlanche, 2023-06-19:
@NevynUK The Meadow.CLI code that sends the command is as follow....

https://github.com/WildernessLabs/Meadow.CLI/blob/develop/Meadow.CLI.Core/Devices/MeadowLocalDevice.FileManager.cs#L429-L452

We're encoding the filename as UTF-8, before sending the command. Would that cause a problem?

NevynUK, 2023-09-18:
This appears to be working in `meadow` version 1.3.0.

closed 2023-09-18 by NevynUK