WildernessLabs/Meadow_Issues #326: JSON Deserializing Bug with External Library Classes

jeffjulian, 2023-06-22:
**Describe the bug**
When an external library is used to deserialize JSON, it throws an error.  This happens with the System.Text.Json and NewtonSoft.Json packages.

**To Reproduce**
Steps to reproduce the behavior:
1. Sample Code Added, just deploy and the error will be logged via the resolver.

**Developer tools (please complete the following information as best as you can):**
Visual Studio 2022, Meadow CLI

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v1
 - Meadow OS: 1.0.0.2

**Additional context**
[MeadowBug.JsonIssue.zip](https://github.com/WildernessLabs/Meadow_Issues/files/11839608/MeadowBug.JsonIssue.zip)

wickism, 2023-06-26:
I'm having a similar issue serializing where library classes serialize to an empty string.

To Reproduce
Steps to reproduce the behavior:
Sample Code Added, just deploy and the error will be logged via the resolver.

Developer tools (please complete the following information as best as you can):
Visual Studio 2022 Preview, Meadow CLI

Meadow (please complete the following information as best as you can):
The attached code will serialize a class from a library will be an empty JSON string.   This code will also serialize the same object located in the meadow app.

Meadow hardware version: F7FeatherV2
Meadow OS: 1.0.2.0

--nolink gets me around this for now, I didn't expect it to strip my own libraries.

[MeadowSerialize.zip](https://github.com/WildernessLabs/Meadow_Issues/files/11874298/MeadowSerialize.zip)

bryancostanich, 2023-10-17:
This appears to be a linking issue with the the linker stripping out key code. @ctacke to document workaround.

DarkNoir, 2023-10-23:
The work around is to deploy via command line with --nolink and not via the VS menu 'Deploy'? Or can this be configured in VS to use when using the menu item 'Deploy'

DarkNoir, 2023-11-10:
I was able to resolve this in .Net 7 using the **DynamicDependencyAttribute** where trimming has the same effect. I do not see a way of doing this in .NET standard 2.1 though.

bryancostanich, 2023-11-10:
oooh! @ctacke @alexischr @adrianstevens  that's probably the replacement to the Xamarin `[Preserve]` attribute we need to support.

DarkNoir, 2023-11-10:
Basically this:
`[DynamicDependency(DynamicallyAccessedMemberTypes.All, "SensorVerse.IoT.Messages.SensorDataMessage", "Environment")]`
is able to tell the linker to include the class and where to find it. I put it on the method that was calling deserialize for that class.

jbeenes, 2024-03-07:
Has this been documented already somewhere? Can't seem to find how to use the workaround.

ctacke, 2024-12-12:
@patridge We need to get this documented and closed

patridge, 2024-12-17:
@ctacke Do you have any experience using that attribute successfully and have an example I can follow along with? I can't validate that it works at all. When I try to add it to a Meadow app, it complains that it can't find it. The [official docs for that attribute](https://learn.microsoft.com/en-us/dotnet/api/system.diagnostics.codeanalysis.dynamicdependencyattribute?view=net-8.0) are mostly worthless, though a [separate page](https://learn.microsoft.com/en-us/dotnet/core/deploying/trimming/prepare-libraries-for-trimming#dynamicdependency) about trimming suggests it's a "last resort".

closed 2025-03-18 by ctacke