WildernessLabs/Meadow_Issues #330: meadow.config.yaml template has incorrect indentation

jorgedevs, 2023-07-12:
When creating a Meadow app project from scratch, open meadow.config.yaml, and uncomment all the fields

`# Uncommented these options as needed.
# To learn more about these config options, check out the OS & Device Configuration documentation.
# http://developer.wildernesslabs.co/Meadow/Meadow.OS/Configuration/OS_Device_Configuration/

Device:
    # Name of the device on the network.
    Name: MeadowF7V2_ConfigSample

    # Uncomment if SD card hardware present on this hardware (e.g., Core-Compute module with SD add-on)? Optional; default value is `false`.
    SdStorageSupported: true
# Control how the ESP coprocessor will start and operate.
Coprocessor:
    # Should the ESP32 automatically attempt to connect to an access point at startup?
    # If set to true, wifi.config.yaml credentials must be stored in the device.
    AutomaticallyStartNetwork: true

    # Should the ESP32 automatically reconnect to the configured access point?
    AutomaticallyReconnect: true

    # Maximum number of retry attempts for connections etc. before an error code is returned.
    MaximumRetryCount: 7
# Network configuration.
Network:
  Interfaces:
      - Name: Ethernet
        UseDHCP: false
        IPAddress: 192.168.1.60
        NetMask: 255.255.255.0
        Gateway: 192.168.1.254
      - Name: WiFi
        UseDHCP: true
        IPAddress:
        NetMask:
        Gateway:

    # Which interface should be used?
    DefaultInterface: WiFi

    # Automatically attempt to get the time at startup?
    GetNetworkTimeAtStartup: true

    # Time synchronization period in seconds.
    NtpRefreshPeriodSeconds: 600

    # Name of the NTP servers.
    NtpServers:
        - 0.pool.ntp.org
        - 1.pool.ntp.org
        - 2.pool.ntp.org
        - 3.pool.ntp.org

    # IP addresses of the DNS servers.
    DnsServers:
        - 1.1.1.1
        - 8.8.8.8`

Network -> Interfaces indentation is wrong so the yaml will give a syntax error.

CartBlanche, 2023-07-17:
@jorgedevs doesn't this PR https://github.com/WildernessLabs/Meadow.Sdk/pull/42/ fix this ?

patridge, 2023-07-17:
@CartBlanche I believe it should, whenever it is deployed. (At least, that was the intention when I adjusted things.)

closed 2023-09-19 by lamebrain