WildernessLabs/Meadow_Issues #335: Enabling SD card on a non-SD platform causes failure to boot

ctacke, 2023-07-22:
Accidentally enabling the SD card on a platform (not sure if it is Feather on Project Lab) caused the platform to not boot at all.  It hangs, probably during SD initialization.  Need to investigate where the hang happens (OS or Core) and see if it's possible to at least give an error or warning to the user.  Best case would be an error in the output and still boot.

Peter-Moody, 2023-07-22:
This is to be expected and why I ask @Mark to add the SD card option to the configuration file.
The hardware only allows the software to detect if it's a FeatherV1, FeatherV2 or CCM. There's no way to determine if the platform can support an SD card. Therefore, you must have tested this on a CCM powered Project Lab setup.

Peter-Moody, 2023-12-19:
After investigating with a Project Lab v3.e found that the problem appears not be a `Meadow.OS` problem.
Having a new PL with the original demo code installed. I added the last 2 lines to the PL's `meadow.config.yaml` file. No other changes were made.
```
MonoControl:
    Options: --jit
Device:
    SdStorageSupported: true
```
Before adding the above 2 lines I could see the demo display. After adding the above 2 lines, the screen remained blank. However, from the CLI, I could issue commands (e.g. `meadow device info`) and it worked perfectly. Which indicates that the  C# application is failing to use the `meadow.config.yaml` information available. It could detect that the SDCard (or Ethernet or Cell) were enabled and provide an appropriate warning, instead of attempting to run.