WildernessLabs/Meadow_Issues #339: App Error in App Run: Cannot clear until Idle

doingnz, 2023-07-30:
**Describe the bug**
During OTA enabled application startup, UpdateService calls ClearUpdates() too soon and an exception terminates the Application start up. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create default New Project 
2. Add OTA support as described here: [http://developer.wildernesslabs.co/Meadow/Meadow.OS/Updates/](http://developer.wildernesslabs.co/Meadow/Meadow.OS/Updates/ )
3. Publish the application to Meadow
4. Reset Meadow and run `meadow listen`  to monitor stdout. 
5. May be necessary to try a few times till it fails

**Expected behavior**
Application should always start and register handlers for UpdateService. 

**Screenshots**
Example of exception error report.
```
Meadow StdOut: Running App
Meadow StdOut: -----------------
Meadow StdOut: Run... Test OtA NORM
Meadow StdOut: -----------------
Meadow StdOut: App Error in App Run: Cannot clear until Idle
Meadow StdOut: System.Exception: Cannot clear until Idle
Meadow StdOut: Interface: wlan0: wlan0 Wireless80211 Up
Meadow StdOut: at Meadow.Update.UpdateService.ClearUpdates () [0x00017] in <53f7bd8df00948368864f54c51d7af47>:0
Meadow StdOut: at MeadowBlinky.MeadowApp.Run () [0x00041] in <69804771ccd0433895c8ab4c0b2cb605>:0
Meadow StdOut: at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00174] in <53f7bd8df00948368864f54c51d7af47>:0
Meadow StdOut: App shutting down
Meadow StdOut: Shutdown
```

**Developer tools:**
 - Microsoft Windows [Version 10.0.22621.1992]
 - Visual Studio Enterprise 2022 [Version v17.6.5]
 - Meadow extension for IDE version: [VS 2022 Tools for Meadow v1.1.0]

**Meadow:**
```
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowCompute_Clima

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 1.1.0.0
    Mono: 1.1.0.0
    Coprocessor: 1.1.0.0
    Protocol: 7
```

**Additional context**
Reset Meadow and it usually runs fine.

doingnz, 2023-08-01:
As a workaround, I added a while loop prior to calling ClearUpdates() to check the State. 

BTW, I am not clear on what the purpose of the call to svc.ClearUpdates() is and consequences if it is not called during startup?

```            var svc = Resolver.Services.Get<IUpdateService>() as Meadow.Update.UpdateService;
            while (svc.State != UpdateState.Dead && svc.State != UpdateState.Idle)
            {
                Resolver.Log.Trace($"svc.State = {svc.State}");
                await Task.Delay(1000);
            }
            svc.ClearUpdates(); // uncomment to clear persisted info

            svc.OnUpdateAvailable += (updateService, info) =>
            {
                Resolver.Log.Info("Update available!");

                // queue it for retreival "later"
                Task.Run(async () =>
                {
                    await Task.Delay(5000);
                    updateService.RetrieveUpdate(info);
                });
            };

            svc.OnUpdateRetrieved += (updateService, info) =>
            {
                Resolver.Log.Info("Update retrieved!");

                Task.Run(async () =>
                {
                    await Task.Delay(5000);
                    updateService.ApplyUpdate(info);
                });
            };
```

ctacke, 2024-09-17:
`ClearUpdates()` deletes all local (on the meadow) information about known updates.  Meadow keeps a log of available, downloaded and applied items to allow a device to not attempt to duplicate downloads or apply updates that have already been applied.  Clearing the updates is generally just a sample app feature so that the device will react every time a package is published.  Without clearing it, the package availability notice will only happen in an app once.

closed 2024-09-17 by ctacke