WildernessLabs/Meadow_Issues #340: VS2022 Deploy/Debug creates App.exe and pushes it to Meadow when App.dll expected

doingnz, 2023-08-01:
**Describe the bug**
Meadow project build correctly creates App.dll as the project output type is `<OutputType>Library</OutputType>`. However, when the same project is loaded into Meadow via Deploy or Debug, the `App.dll` is copied to `App.exe` and this `App.exe` is what is actually deployed into the Meadow file system. This triggers other issues later when OTA Publish only pushes 'App.dll`

**To Reproduce**
Steps to reproduce the behavior:
1. Project New to create Blinky sample app
2. Delete the project bin folder if present
3. Build the project and note App.dll created and there is no App.exe
4. Deploy or Debug the project and note App.exe is created. 
5. Use `meadow file list` to confirm App.exe was added to Meadow 

**Expected behavior**
No `App.exe` to be created during Deploy or Debug launch of the Meadow application. Deploy to Meadow should upload `App.dll` only. 

**Developer tools :**
- Microsoft Windows [Version 10.0.22621.1992]
- Visual Studio Enterprise 2022 [Version v17.6.5]
- Meadow extension for IDE version: [VS 2022 Tools for Meadow v1.1.0]

**Meadow :**
```Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowCompute_Clima

Hardware Information
    Processor type: STM32F777IIK6
    ID: 22-00-26-00-11-51-36-30-33-33-37-33
    Serial number: 335D335F3036
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F4:0C

Firmware Versions
    OS: 1.1.0.0
    Mono: 1.1.0.0
    Coprocessor: 1.1.0.0
    Protocol: 7]
```
**Additional context**
Example file list after deploy showing `App.exe` present on Meadow file system
```C:\Windows\System32>meadow file list

Connecting to Meadow on COM6

Getting files on partition 0
app.config.yaml                                     730 bytes
App.deps.json                                        25 kB
meadow.config.yaml                                    1 kB
Meadow.deps.json                                     23 kB
Meadow.xml                                           64 kB
App.exe                                              14 kB
Meadow.Contracts.dll                                127 kB
Meadow.dll                                          154 kB
Meadow.F7.dll                                       170 kB
Meadow.Foundation.dll                                19 kB
Meadow.Logging.dll                                    7 kB
Meadow.Logging.Udp.dll                                5 kB
Meadow.Units.dll                                     98 kB
Microsoft.Bcl.AsyncInterfaces.dll                     5 kB
Mono.Security.dll                                   119 kB
MQTTnet.dll                                         292 kB
mscorlib.dll                                        4.6 MB
netstandard.dll                                      56 kB
System.Buffers.dll                                    4 kB
System.Configuration.dll                             51 kB
System.Core.dll                                      50 kB
System.dll                                          867 kB
System.IO.Compression.dll                            76 kB
System.IO.Compression.FileSystem.dll                  7 kB
System.Memory.dll                                     4 kB
System.Net.Http.dll                                 204 kB
System.Numerics.dll                                  25 kB
System.Numerics.Vectors.dll                           4 kB
System.Runtime.CompilerServices.Unsafe.dll            2 kB
System.Text.Encodings.Web.dll                        35 kB
System.Text.Json.dll                                396 kB
System.Threading.Tasks.Extensions.dll                 5 kB
System.Web.dll                                       97 kB
System.Xml.dll                                      694 kB
App.pdb                                              11 kB
Meadow.Logging.Udp.pdb                                8 kB
Meadow.pdb                                           50 kB
mscorlib.pdb                                        1.7 MB
System.Core.pdb                                      25 kB
System.pdb                                          347 kB
meadow.log                                           97 bytes
dns.conf                                             19 bytes

Summary:
        42 files
        10.40MB of file data
        Spanning 2562 blocks
```

CartBlanche, 2023-09-05:
@lamebrain I have a PR with a fix for this. Hopefully have this closed next week.

CartBlanche, 2023-09-26:
Should be fixed in 1.3.4 Release.

closed 2023-09-26 by CartBlanche